open-mmlab/OpenPCDet #571: How can I display the model predictions and GroundTruth simultaneously?

TyroneLi, 2021-06-21:
After training, I wanna display the predictions and the GT using demo.py. Currently, I notice that demo.py could only visualize the predictions, how can I make adjustment to display the GT simultaneously. Thanks.

jihanyang, 2021-06-21:
Hello, you can refer to this function
https://github.com/open-mmlab/OpenPCDet/blob/26a16123cf820e29ed565f4ad53964c66f3ea030/tools/visual_utils/visualize_utils.py#L142

TyroneLi, 2021-06-21:
> Hello, you can refer to this function
> https://github.com/open-mmlab/OpenPCDet/blob/26a16123cf820e29ed565f4ad53964c66f3ea030/tools/visual_utils/visualize_utils.py#L142

Ok, I have to provide gt_boxes for data_dict and normalize it corresponding to prediction boxes coordinate. The I expect when passing it to draw_scenes and obtain my expectation.

Manueljohnson063, 2021-06-22:
Hi @jihanyang @TyroneLi @sshaoshuai 
I am currently using my own dataset for training the network. My lidar is place at a height of 4.89meters from the ground .
after reading the comments (issues) i notice that if we are using our own dataset we need to translate and rotate the data and make it in the form of kitty .But due to some problem i am not able to translate my point cloud to 1.73 meters (because of data loss).all other things I have done in the kitty format.
Could you please tell am I going in the correct way? Here I am using only point cloud not images.

jihanyang, 2021-06-23:
@Manueljohnson063 May I know why you cannot translate your point cloud?

Manueljohnson063, 2021-06-23:
@jihanyang 
Many  many thanks for your immediate replay
About my point cloud,
1.**I am using 32 layer lidar point cloud**
2**.My ground is not plane it is terrain**
My lidar is 4.96m from the ground ,objects I want to detect are 
1.Human,Rock,trees
The above objects will be either placed in **pit or on surface** so if I translate the PC I thought that  I loose the data of objects in the pit.

jihanyang, 2021-06-23:
@Manueljohnson063 Hello, I am not well understanding what are the issues caused by point cloud translating. However, you can refer to our another repo: https://github.com/CVMI-Lab/ST3D/issues/5. Since different datasets have different LiDAR hight from the ground, we just translate the point cloud to the same origin points in the z-axis. It seems that this simple translation works without problems.

zjx99, 2021-08-09:
> Hello, you can refer to this function
> https://github.com/open-mmlab/OpenPCDet/blob/26a16123cf820e29ed565f4ad53964c66f3ea030/tools/visual_utils/visualize_utils.py#L142

Hi, jihanyang,

I remember you commented somewhere that you saved the point clouds and the predicted boxes to visualize on local machine with mayavi. Can you please share the codes Thanks.

weijiawang96, 2021-10-02:
> Ok, I have to provide gt_boxes for data_dict and normalize it corresponding to prediction boxes coordinate. The I expect when passing it to draw_scenes and obtain my expectation.

Hi, it seems that unlike what we do in train.py (where loaded data has ground truths), data_dict in demo.py does not have the key of "gt_boxes". How can we change that? Thanks

closed 2022-03-08 by github-actions[bot]