open-mmlab/OpenPCDet #573: pcded ---> AttributeError: 'PosixPath' object has no attribute 'flush

KangChou, 2021-06-25:
`2021-06-25 09:23:23,992   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2021-06-25 09:23:24,091   INFO  Total number of samples: 	1
2021-06-25 09:23:26,637   INFO  ==> Loading parameters from checkpoint ckpt/pv_rcnn_8369.pth to CPU
2021-06-25 09:23:27,441   INFO  ==> Done (loaded 367/367)
Traceback (most recent call last):
  File "demo.py", line 103, in <module>
    main()
  File "demo.py", line 87, in main
    for idx, data_dict in enumerate(demo_dataset):
  File "demo.py", line 41, in __getitem__
    points = np.fromfile(self.sample_file_list[index], dtype=np.float32).reshape(-1, 4)
AttributeError: 'PosixPath' object has no attribute 'flush'
`

KangChou, 2021-06-26:
The problem has been solved.
The main reason is the conflict between different versions of QT

Molocks, 2021-07-20:
> The problem has been solved.
> The main reason is the conflict between different versions of QT

I also met the same problem. Could you please tell me how you  solved it？

closed 2022-03-08 by github-actions[bot]