open-mmlab/OpenPCDet #574: How to get evalution result of each class?

yaxi333, 2021-06-25:
Hi 

I have noticed that the model zoo result which is shown on the website:
[image]
I want to ask how you get this Car@R11? Because it seems to be two AP results based on different thresholds 0.7 and 0.5 for moderate 3d detection task @R11? What I have got is as following:
Car AP@0.70, 0.70, 0.70:
bbox AP:94.7996, 88.9117, 87.8464
bev AP:89.7246, 86.8356, 83.5177
3d AP:86.4783, 76.2949, 72.4491
aos AP:94.75, 88.76, 87.63
Car AP@0.70, 0.50, 0.50:
bbox AP:94.7996, 88.9117, 87.8464
bev AP:94.9225, 89.3233, 88.8077
3d AP:94.8194, 89.2578, 88.6416
aos AP:94.75, 88.76, 87.63

Thanks in advance!

jihanyang, 2021-06-26:
The results on IoU threshold = 0.7 are reported.

yaxi333, 2021-06-26:
Thank you very much! Then what the threshold reported for Pedestrian and Cyclist?

jihanyang, 2021-06-27:
both 0.5

closed 2022-03-08 by github-actions[bot]