open-mmlab/OpenPCDet #575: Successfully Installed in the Colab

zzmonlyyou, 2021-06-27:
Hi,

Thank you for your contribution. I tried to install the OpenPCDet in the Colab. I first installed the spconv 1.2.1. Then I installed the OpenPCDet 0.3.0. 

I modified the demo.py and named it as test_2.py so it will only save the results install of showing it (since colab does not support mayavi).
Everything works fine on the local machine.

But when I run the modified demo.py, here is what I got:
Traceback (most recent call last):
  File "test_2.py", line 11, in <module>
    from pcdet.models import build_network, load_data_to_gpu
  File "/content/OpenPCDet/pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/content/OpenPCDet/pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/content/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 7, in <module>
    from .. import backbones_2d, backbones_3d, dense_heads, roi_heads
  File "/content/OpenPCDet/pcdet/models/backbones_3d/__init__.py", line 2, in <module>
    from .spconv_backbone import VoxelBackBone8x, VoxelResBackBone8x
  File "/content/OpenPCDet/pcdet/models/backbones_3d/spconv_backbone.py", line 3, in <module>
    import spconv
  File "/usr/local/lib/python3.7/dist-packages/spconv-1.2.1-py3.7-linux-x86_64.egg/spconv/__init__.py", line 35, in <module>
    torch.ops.load_library(_LIB_PATH)
  File "/usr/local/lib/python3.7/dist-packages/torch/_ops.py", line 106, in load_library
    ctypes.CDLL(path)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libtorch_cpu.so: cannot open shared object file: No such file or directory

Thank you for you help!

jihanyang, 2021-06-27:
Can you import spconv successfully?

zzmonlyyou, 2021-06-28:
> Can you import spconv successfully?

Yes, I can import spconv.

brankyy, 2021-07-15:
Have you solved this problem?
I'd like to run this freamwork on Colab too. Did you try to use open3d instead of mayavi on Colab ?

brankyy, 2021-07-28:
> Have you solved this problem?
> I'd like to run this freamwork on Colab too. Did you try to use open3d instead of mayavi on Colab ?

I tried to use open3d on Colab, but it didn't work. So I choose to save the result instead of showing it like @zzmonlyyou mentioned.

Simply comment Line 5, 13, 93-97. Save data_dict and  pred_dict as pkl. Download it and visualize the result locally.

closed 2022-03-08 by github-actions[bot]