open-mmlab/OpenPCDet #579: Used my own data, but it outputs no results

fys445089968, 2021-07-02:
In the quick demo, I used a frame of point cloud I collected by myself, and its format is .bin. I use kitti_models/pv_rcnn.yaml, but there is no prediction result after input. How can I modify the cfg file to adapt to my own data.Look forward to your answer.

thptai, 2021-07-02:
You should check the value of your point cloud coordinates.

For example, KITTI cfg defines `POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]`

The setting is based on meter. So if your point cloud is based on centimeter, the detection range is very small and probably can not find any object inside that area.

HuangVictorAuto, 2021-07-20:
@fys445089968 , I have also had the same problem. I tried to use a pretrained model with my own data for a quick demo. It loaded the point cloud, but no prediciton box. Have you figured it out? 
[image]

closed 2022-03-08 by github-actions[bot]

OrangeSodahub, 2022-07-26:
Hello, you can refer to my successful example using kitti format custom dataset. README describes how to label, train, inference it including transformation of coordinates. It may solve your problems!
https://github.com/OrangeSodahub/CRLFnet#lid-cam-fusion
https://github.com/OrangeSodahub/CRLFnet/blob/master/src/site_model/src/LidCamFusion/OpenPCDet/pcdet/datasets/custom/README.md