open-mmlab/OpenPCDet #581: Reproducability of CaDDN

MartinHahner, 2021-07-07:
@codyreading 

Unfortunately, I was not able to reproduce your CaDDN results for the **Cyclist and Pedestrian** class.
Shown here are four runs with the provided config and the average over those runs given in the first line:
[image]
The columns of the individual class results from left to right are always _easy - moderate - hard_ (R40).

[Here](https://wandb.ai/martin_hahner/cvpr_2022/reports/CaDDN--Vmlldzo4MzMzODU?accessToken=yv5srcmfjx2poyqn1mpno8lyfhfxiow9v2ev88qd4mgmgsyn5h4g9nclgfpahmi4) you can maybe find some more details.

codyreading, 2021-07-07:
Hi Martin!

Quick question, are you always reporting result at Epoch 80? I would recommend selecting the epoch with the highest results, which in my experience has ranged between Epochs 60-80. My strategy was always to select the epoch with the highest Car3d_R40 moderate.

codyreading, 2021-07-07:
The results you have shown seem pretty typical for most of my runs, I only selected the best model to save in this repo.

MartinHahner, 2021-07-08:
Hi Cody, 
thanks for your swift reply.

> Quick question, are you always reporting results at Epoch 80? I would recommend selecting the epoch with the highest results, which in my experience has ranged between Epochs 60-80.

No, the results are from the epoch (indicated in the column just left of the results) with the highest score. 
So in the screenshot of my experiments, the best epoch varies between Epochs 60 and 73.

>  My strategy was always to select the epoch with the highest Car3d_R40 moderate.

My strategy to select the best epoch is exactly like yours (the main metric of KITTI).

Are you able to reproduce the results in the current version of this "official" OpenPCDet repo (or only in your fork)?
[I am **not** doubting that one can reproduce the results in your fork.]

codyreading, 2021-07-08:
Hi @MartinHahner,

Yes, I can confirm I have reproduced similar results between my fork and the "official" OpenPCDet repo.

I think the issue here is that we are comparing your results to my best model, rather than the average over the runs. Results on classes with smaller numbers such as Pedestrian and Cyclist have quite a bit of variance with CaDDN, especially on the small validation set where getting a few extra boxes correct greatly affects the AP results. Rather, I've included the same format of results that you have shown here when training on this repo:

[image]

Car and Cyclist average results look similar to what you report. However, your Pedestrian results are decently higher than mine. This could be that you got lucky with your runs, or the fact that you are using a larger batch size (Looks like you are using 4 TITAN RTXs, so likely your batch size is quite large relative that what I was using (batch_size of 2)) which could have affect results.

Hope this helps!

MartinHahner, 2021-07-08:
> [image]

OK, so your and my results match, 
but the performance given in the [table](https://github.com/open-mmlab/OpenPCDet#kitti-3d-object-detection-baselines) of the readme, especially for the **Cyclist** class is much higher.

From the table in the readme:
|Car | Pedestrian | Cyclist|
-- | -- | --
|21.38 | 13.02 | **9.76** |

If I understand you correctly, you attribute this difference to the following reason, and if one gets lucky, one can indeed achieve this performance mentioned in the readme.

> especially on the small validation set where getting a few extra boxes correct greatly affects the AP results.

Still seems a bit odd to me.

codyreading, 2021-07-08:
So the table of the README is AP@R11, which is known to have flaws in its evaluation. I would've included the R40 results in the README but didn't want to reformat the whole table.

MartinHahner, 2021-07-09:
I know that the results in the readme are AP@11, but did not know that the difference to AP@40 could be so significant.
I rooted to change the results in the readme to AP@40 a long time ago (@sshaoshuai @jihanyang).
One should not report AP@11 anymore.

Thanks for your clarifications @codyreading.

closed 2021-07-09 by MartinHahner

codyreading, 2021-07-09:
My pleasure, happy to clarify.

The main issue with R11 is exactly this, as discussed in [MonoDIS](https://arxiv.org/pdf/1905.12365.pdf). Essentially, since R11 includes a recall step of 0 while R40 does not, it is incredibly easy to get 100% precision at the lowest recall step (recall = 0), which gives a free roughly 9.09% AP. Essentially, this causes large difference between R11 and R40 when detection performance is poor, as AP@R40 will often be close to 0 while R11 will be around 9%.

MartinHahner, 2021-07-09:
Shoutout to @simonelli-andrea

jihanyang, 2021-07-10:
@MartinHahner Hello, thanks for your advice, I also observe that AP@R11 and AP@R40 could be really different when they are low. Maybe we could add another AP@R40 table in readme, since a lot of paper still use the reuslts of AP@R11, we may still keep it.