open-mmlab/OpenPCDet #582: cannot download pretrained model

clytze0216, 2021-07-08:
hi @lindahua @sshaoshuai @hellock @MMLAB-CUHK 
thanks for your great work!
but I have no VPN to download the pre-trained models on drivegoogle website.
Could you offer other ways to download ?(I'm in china)
It means a lot to me .Looking forward to your reply.
Thanks very much.

brankyy, 2021-07-14:
if you will,  you could leave an email address and the pretrained model you need, I can download it and send it to you :)

jiaoluonanhai, 2021-09-30:
Could you send me a copy ? my email is 1054953459@qq.com. Thanks very much. @brankyy

applededipan, 2021-12-13:
Could you please send me a copy ? I tried many times but still can not download! my email is 524888133@qq.com. Thank you very much!@brankyy

Typolo, 2021-12-21:
@brankyy could please send me the pointpillar model (model-18m), my email is 3026462963@qq.com, thank you so much.

155cannon, 2021-12-24:
I also need  the pointpillar model (model-18m), thank you!@brankyy

155cannon, 2021-12-24:
I also need the pointpillar model (model-18m), thank you!@brankyy   my e-mail:155cannon@163.com

closed 2022-03-08 by github-actions[bot]