open-mmlab/OpenPCDet #584: no 2d boxes data. how to test my own data.

clytze0216, 2021-07-09:
hi@open-mmlab team 
I built my own data set for training.I don't have 2D Box data in my dataset, so I've annotated all the Box code in kitti_data.py.You can train normally.
But when I run test.py, I get a lot of errors.The main error occurs in eval.Since I annotated all the code about Box when preparing the data, when I wanted to output AP, I could not get the value of Box, so I could not calculate IOU and other values.I wonder how I can solve this problem.Thank you in advance and look forward to your reply. Thank you!!!!!!!!!!

Manueljohnson063, 2021-08-15:
Hi @clytze0216 ,
Have you done it with your own data??????if so can u give some advice.......
Thankz in advance

jihanyang, 2021-08-16:
You can follow here https://github.com/open-mmlab/OpenPCDet/blob/8e055f02df245b1f0675e874792dd708fb84c04b/pcdet/datasets/waymo/waymo_dataset.py#L203. 2D box are not necessary.

closed 2022-03-08 by github-actions[bot]