open-mmlab/OpenPCDet #589: The difference of Voxelrcnn

chenxyyy, 2021-07-14:
Hi @sshaoshuai .

I noticed that the voxelrcnn in Openpcdet is a little different from the official implementation.

1.  theBACKBONE_2D layer num. your implementation is [5, 5], but official implementation is [4, 4]
2.  the ROI_GRID_POOL. your implementation is FEATURES_SOURCE: ['x_conv2', 'x_conv3', 'x_conv4']. but official implementation is ['x_conv3', 'x_conv4']. 

I want to know why do you make these changes.

jihanyang, 2021-07-15:
@djiajunustc

djiajunustc, 2021-07-15:
Hi @chenxyyy,

The codes of Voxel R-CNN in this repository are also implemented by me.
Voxel R-CNN is not sensitive to these trivial differences.
Thanks for your attention.

Best Regards,
Jiajun Deng

chenxyyy, 2021-07-16:
Okay, okay, I get it. Thank you very much.

While achieving better results, it is also very fast.

closed 2021-09-16 by jihanyang