open-mmlab/OpenPCDet #590: Can these models be implemented to a Lidar with 32 lines?

YunxiangCai, 2021-07-14:
Hi, these models are trained with Lidar of 64 lines. Can these pre-trained models be used for Lidar with 32 lines?

jihanyang, 2021-07-14:
Sure, but the results might not be great enough. You can refer to domain adaptation works such as our ST3D https://github.com/CVMI-Lab/ST3D

FlyerMaxwell, 2021-07-14:
> Sure, but the results might not be great enough. You can refer to domain adaptation works such as our ST3D https://github.com/CVMI-Lab/ST3D

Thanks for your kindly reply.

 Actually, I am pretty new to the lidar tracking problem. My task is to implement a model to track one specific car with my own lidar of 32 lines to be a ground truth of another wireless sensing system. 

My own dataset is not large. So firstly I want to try a pre-trained SOTA model with my own data. Could you kindly give me more hint to do this project?

Thanks in advance!

jihanyang, 2021-07-15:
Hello, I am also not experienced on tracking problem. Pretraining and finetuning is a good pipeline for a new dataset with only small number of samples. I recommond to consider the large dataset such as waymo pretrained model if you just consider single frame detection with tracking method such as AB3DMOT. If you want to use multi-sweeps fusion to improve the performacne of detection on your dataset, you can select the fused nuScenes pre-trained model. Just some trival thoughts. Hope can help you.

FlyerMaxwell, 2021-07-15:
> Hello, I am also not experienced on tracking problem. Pretraining and finetuning is a good pipeline for a new dataset with only small number of samples. I recommond to consider the large dataset [...]

Thanks a million for your great advice.
I have transformed the data collected by RoboSense（32 lines）to KITTI format and tried the pre-trained models. The result seems pretty good. Does mmlab provide a tracking model you mentioned? I cannot find it.

jihanyang, 2021-07-16:
Sorry, as I know, there is no tracking model provided in openmmlab.

FlyerMaxwell, 2021-07-16:
> Sorry, as I know, there is no tracking model provided in openmmlab.

Thanks again.

closed 2021-07-16 by FlyerMaxwell

DavidWZX, 2021-12-27:
> Thanks a million for your great advice. I have transformed the data collected by RoboSense（32 lines）to KITTI format and tried the pre-trained models. The result seems pretty good. Does mmlab provide a tracking model you mentioned? I cannot find it.

Hello, @jihanyang  I'm struggling with testing on my own dataset which recorded by robosense(32 beams). I just noticed that you have done this kind of work, right? Can you tell me which pre-trained model you used? Much thanks!

wayne1627, 2022-03-06:
> Thanks a million for your great advice. I have transformed the data collected by RoboSense（32 lines）to KITTI format and tried the pre-trained models. The result seems pretty good. Does mmlab provide a tracking model you mentioned? I cannot find it.

Did you set the heigt as 1.73m(which just same as kitti dataset ) to record your own dataset(32 lines), and use a pretrained kitti model to inference on your own dataset, and it works well, right?

wayne1627, 2022-03-06:
> > Thanks a million for your great advice. I have transformed the data collected by RoboSense（32 lines）to KITTI format and tried the pre-trained models. The result seems pretty good. Does mmlab provide a tracking model you mentioned? I cannot find it.
> 
> Hello, @FlyerMaxwell ! I'm struggling with testing on my own dataset which recorded by robosense(32 beams). I just noticed that you have done this kind of work, right? Can you tell me which pre-trained model you used? Much thanks!

Did you tried this work? Does it works well? hoping for your reply