open-mmlab/OpenPCDet #592: Questions about cfg file of pv_rcnn

gaojunbin, 2021-07-14:
Hi, @sshaoshuai,
Hi, all,

I am confused for some difference between the paper and code.

In your PV RCNN paper, it is written that:

> For inference, we keep the top-100 proposals generated
> from the 3D voxel CNN with a 3D IoU threshold of 0.7 for non-maximum-suppression (NMS). These proposals are further refined in the proposal refinement stage with aggre- gated keypoint features. We finally use an NMS threshold of 0.01 to remove the redundant boxes.

Paper reports that nms threshold is 0.01 to remove the redundant boxes.  However, I find in the config file: pv_rcnn.yaml, it is 0.1 in [pv_rcnn.yaml/POST_PROCESSING/NMS_CONFIG/NMS_THRESH.](https://github.com/open-mmlab/OpenPCDet/blob/63d6540ca692d62fe5447bfe7c63233126170774/tools/cfgs/kitti_models/pv_rcnn.yaml#L216)

I found the number is 0.01 in 'pointpillar.yaml' and some others. 

Is there something wrong or I got it wrong? Thx, sorry for troubling.

jihanyang, 2021-07-15:
Hello, you can try to modify it on the provided pre-trained model. Due to the sparsity of point clouds, I guess the final results are quick similiar. Notice that the NMS_THRESH for second is 0.01.

gaojunbin, 2021-07-15:
> Hello, you can try to modify it on the provided pre-trained model. Due to the sparsity of point clouds, I guess the final results are quick similiar. Notice that the NMS_THRESH for second is 0.01.

Thanks for your reply! I will try it.

closed 2021-07-15 by gaojunbin