open-mmlab/OpenPCDet #596: change the point range and voxel size of SECOND. the AP of pedestrian was very low

clytze0216, 2021-07-21:
Hello, @open-mmlab
I want to train my own dataset . In order to adapt to the point cloud range of my own dataset, I modified the point range in kitti_dataset.yaml as:[-422.4,-440,-3,422.4,440,1], voxel size as [0.6,0.5,0.1].
but when I test the AP . I got the result:
the AP of pedestrian is only nearly 10.
the AP odf the car is correct ,nearly 70.
Could you please tell me if there is any problem if I modify the point range and voxel size according to the above scope?
Why do pedestrians  get poor test results?
Is there anything you can do to improve pedestrian detection?Looking forward to your reply!Thank you very much!

jihanyang, 2021-07-21:
voxel size is too large for small categories like pedestrian

clytze0216, 2021-07-22:
Thanks a lot!!!!!
One more question: My own dataset point cloud is very large, and if I tried to keep the same voxel size, I would generate too many voxels, which would exceed the number specified in the code, and would result in an error.  So I would like to ask you if I want to keep a small voxel size so that it can detect pedestrians and cyclist, but I also want to expand the point range, what should I do?  Hope to get your help, thank you very much!

jihanyang, 2021-07-22:
We don't have experience on LiDAR with such large range. Maybe crop the point cloud will help.

closed 2022-03-08 by github-actions[bot]