open-mmlab/OpenPCDet #597: Run test.py on Kitti test set, AP are all zero?

kyrie-H, 2021-07-21:
In project pv-rcnn,when I run test.py on Kitti test set,the values of recall are zero,I think I haven't configured it well.
[image]
I only set the python file kitti_dataset.py like that:
[image]
How can I do the right test?
please give me a hand!best whishes!

clytze0216, 2021-07-21:
@kyrie-H
 I met the same error.

jihanyang, 2021-07-21:
This does not stands for errors.  Gts are not given in the test set, so these metrics can not be calculated. You need to submit to the kitti website to obtain results.

kyrie-H, 2021-07-21:
> 这并不代表错误。测试集中没有给出 Gts，因此无法计算这些指标。您需要提交到kitti网站才能获得结果。

Thanks a lot！I think you are right,One more question，what should I submit？Are the well-trained pth and the eval result enough?

jihanyang, 2021-07-22:
You should set --save_to_file when inference, and re-zip all those generated .txt files to another zip file.

closed 2021-07-22 by kyrie-H

mqm-slice, 2021-11-22:
What should I do? I tried to use my own program to convert the results in the result.pkl file to the specified format. However, the result was 0 on the 3D index.