open-mmlab/OpenPCDet #598: python train.py error

journey-wang, 2021-07-22:
CUDA_VISIBLE_DEVICES=0,1 python -m torch.distributed.launch train.py --cfg_file cfgs/kitti_models/pointpillar.yaml

2021-07-22 00:43:24,738   INFO  **********************Start training kitti_models/pointpillar(default)**********************
epochs:   0%|                                                                                                                                                                              | 0/80 [00:00<?, ?it/s]/root/OpenPCDet/pcdet/models/dense_heads/target_assigner/axis_aligned_target_assigner.py:167: UserWarning: This overload of nonzero is deprecated:                                       | 0/1979 [00:00<?, ?it/s]
        nonzero()
Consider using one of the following signatures instead:
        nonzero(*, bool as_tuple) (Triggered internally at  /pytorch/torch/csrc/utils/python_arg_parser.cpp:882.)
  fg_inds = (labels > 0).nonzero()[:, 0]
epochs:   0%|                                                                                                                                                                              | 0/80 [00:04<?, ?it/s]
Traceback (most recent call last):
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 153, in main
    train_model(
  File "/root/OpenPCDet/tools/train_utils/train_utils.py", line 86, in train_model
    accumulated_iter = train_one_epoch(
  File "/root/OpenPCDet/tools/train_utils/train_utils.py", line 38, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/root/OpenPCDet/pcdet/models/__init__.py", line 42, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/OpenPCDet/pcdet/models/detectors/pointpillar.py", line 14, in forward
    loss, tb_dict, disp_dict = self.get_training_loss()
  File "/root/OpenPCDet/pcdet/models/detectors/pointpillar.py", line 27, in get_training_loss
    loss_rpn, tb_dict = self.dense_head.get_loss()
  File "/root/OpenPCDet/pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss
    cls_loss, tb_dict = self.get_cls_layer_loss()
  File "/root/OpenPCDet/pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss
    cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights)  # [N, M]
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/OpenPCDet/pcdet/utils/loss_utils.py", line 59, in forward
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
RuntimeError: The size of tensor a (4) must match the size of tensor b (3) at non-singleton dimension 0
                                                                                                                                                                                                                  Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.8/dist-packages/torch/distributed/launch.py", line 260, in <module>
    main()
  File "/usr/local/lib/python3.8/dist-packages/torch/distributed/launch.py", line 255, in main
    raise subprocess.CalledProcessError(returncode=process.returncode,
subprocess.CalledProcessError: Command '['/usr/local/bin/python', '-u', 'train.py', '--local_rank=0', '--cfg_file', 'cfgs/kitti_models/pointpillar.yaml']' returned non-zero exit status 1.

journey-wang, 2021-07-23:
def get_cls_layer_loss(self):
        cls_preds = self.forward_ret_dict['cls_preds']
        box_cls_labels = self.forward_ret_dict['box_cls_labels']
        batch_size = int(cls_preds.shape[0])
        print("\n 000000000000000000 \n")
        print(cls_preds)
        print("\n 1111111111111111111 \n")
        print(batch_size)
        print("\n EEEEEEEEEEEEEEEEEE \n")
        cared = box_cls_labels >= 0  # [N, num_anchors]

After some debugging, I found that sometimes the batch_size here went to 3, not 4.  So please help to provide some advices why batch_size not 4 always?

jihanyang, 2021-07-23:
If the number of samples are not enough to fit the batch in the last iteration, this situation would take place.

journey-wang, 2021-07-23:
> If the number of samples are not enough to fit the batch in the last iteration, this situation would take place.

2021-07-23 05:04:54,448   INFO  Database filter by min points Car: 185273 => 3802
2021-07-23 05:04:54,458   INFO  Database filter by min points Pedestrian: 54566 => 330
2021-07-23 05:04:54,460   INFO  Database filter by min points Cyclist: 5978 => 102
2021-07-23 05:04:54,660   INFO  Loading KITTI dataset
2021-07-23 05:04:54,986   INFO  Total samples for KITTI dataset: 7915

Does my dataset ok?  Thanks very much.

journey-wang, 2021-07-23:
I found that in "pcdet/datasets/processor/data_processor.py"
voxel_generator.generate(points) will return empty result. Please help !!

def transform_points_to_voxels(self, data_dict=None, config=None, voxel_generator=None):
        if data_dict is None:
            try:
                from spconv.utils import VoxelGeneratorV2 as VoxelGenerator
            except:
                from spconv.utils import VoxelGenerator

            voxel_generator = VoxelGenerator(
                voxel_size=config.VOXEL_SIZE,
                point_cloud_range=self.point_cloud_range,
                max_num_points=config.MAX_POINTS_PER_VOXEL,
                max_voxels=config.MAX_NUMBER_OF_VOXELS[self.mode]
            )
            grid_size = (self.point_cloud_range[3:6] - self.point_cloud_range[0:3]) / np.array(config.VOXEL_SIZE)
            self.grid_size = np.round(grid_size).astype(np.int64)
            self.voxel_size = config.VOXEL_SIZE
            return partial(self.transform_points_to_voxels, voxel_generator=voxel_generator)

        points = data_dict['points']
        **voxel_output = voxel_generator.generate(points)**
        if isinstance(voxel_output, dict):
            voxels, coordinates, num_points = \
                voxel_output['voxels'], voxel_output['coordinates'], voxel_output['num_points_per_voxel']

MartinHahner, 2021-07-26:
I think this can also happen if there are no points in the scene (resp. the camera FOV), however, this should not be the case if your KITTI dataset is intact.

SirJamie, 2021-09-27:
Hi @journey-wang , Have you fix this problem?

deepmeng, 2021-11-25:
Hi, man. I have a question. When I train the Second.yaml on KITTI (car, ped. and cyc.),  I found that the values in box_cls_labels of -1, 0, 1, 2 and 3. As far as I know, 0 means the background, and 1, 2, 3 represent the objects' three categories. However, what is the -1 ? @MartinHahner, @jihanyang

deepmeng, 2021-11-25:
> Hi, man. I have a question. When I train the Second.yaml on KITTI (car, ped. and cyc.), I found that the values in box_cls_labels of -1, 0, 1, 2 and 3. As far as I know, 0 means the background, and 1, 2, 3 represent the objects' three categories. However, what is the -1 ? @MartinHahner, @jihanyang

Maybe, '-1' means the 'dont care' category?

closed 2022-03-08 by github-actions[bot]