open-mmlab/OpenPCDet #602: CenterPoint config

SH-Tan, 2021-07-30:
Hello, does this includes the centerpoint densehead code?

MartinHahner, 2021-07-30:
@sshaoshuai @jihanyang 
Yes, it would be great if you could push CenterPoint support to the public repo!

I know you have it internally. :wink: 
https://github.com/open-mmlab/OpenPCDet/pull/549/files#diff-6aa849eac2d13c197e5de529cc91f057dc545f5d9d98ce87e0a63e81f8e00b4c

sshaoshuai, 2021-08-02:
We do have an internally implementation of CenterPoint, but the performance is not matched with official Centerpoint repo yet. It should be easy to add CenterPoint (just a CenterHead) to this repo, but the performance matching may still need some time. 
I am busy these days  and may find time to explore it. (You could also try it by youself if you really need it right now)

MartinHahner, 2021-10-21:
> Any update on when/if CenterPoint is going to be added?

@sidml: It is not so difficult to integrate the config yourself that is mentioned here. 
https://github.com/open-mmlab/OpenPCDet/issues/602#issuecomment-906094762

hz3014, 2021-11-08:
> 
> 
> > Any update on when/if CenterPoint is going to be added?
> 
> @sidml: It is not so difficult to integrate the config yourself that is mentioned here. [#602 (comment)](https://github.com/open-mmlab/OpenPCDet/issues/602#issuecomment-906094762)

Hi Martin, Just wondering did you make it work? I tried to implemented the config, it was training but i got 0 accuracy during evaluation. Not really sure it's the evaluation metrics code or config that i got wrong with.....

MartinHahner, 2021-11-08:
I got it working. I did not change the config. I copied it as is.
Then I just solved one error message after the other (maybe 5 in total), after that, it worked as expected.

hz3014, 2021-11-08:
> 
> 
> I got it working. I did not change the config. I copied it as is. Then I just solved one error message after the other (maybe 5 in total), after that, it worked as expected.

Thanks for your reply. Just wondering is it KITTI or Waymo or both that you got it working?

hz3014, 2021-11-08:
> 
> 
> Neither, I am currently working with [the STF dataset](https://github.com/princeton-computational-imaging/SeeingThroughFog). But I adopted the KITTI dataloader to get STF working, so I am sure it would work with KITTI and assume with Waymo, too.

Thanks for the info. I will try it again.

sshaoshuai, 2021-12-01:
Hi all, 
I have updated OpenPCDet to v0.5.0. This version has supported `CenterHead` and CenterPoint 3D detector. And I guess the performance of CenterPoint is fully reproduced. 
You could have a try.

closed 2021-12-01 by sshaoshuai