open-mmlab/OpenPCDet #605: Some questions about Waymo Open Dataset

Ultra11nstinct, 2021-07-31:
Hi, sorry to bother @Gus-Guo @jihanyang @sshaoshuai @MartinHahner 

I want to know some details about preprocessing of WOD in advande due to the long processing period it takes.
1. i noticed that you use 20% of the whole WOD dataset to train the network, but the Imagesets for WOD seems it did not split the 798 training set, could you provide your split file? or i need to customize one by myself. i also notice that there is a parameter `sampled_interval` in function `create_waymo_infos`, or i just need to change it to 5 to obtain 20% processed_data, which can fit the sampled_interval in `waymo_dataset.yaml`?( i am not sure about this, is that right?)
2. if i want to use same proportion data( e.g. 20%), can i only process the 20% data by using  `create_waymo_infos` function instead of the whole 798 training sequence.

Thanks very much for your contributions.
Looking forward to your reply.

jihanyang, 2021-08-05:
You need to change the sample_interval here: https://github.com/open-mmlab/OpenPCDet/blob/8e055f02df245b1f0675e874792dd708fb84c04b/pcdet/datasets/waymo/waymo_dataset.py#L327

jihanyang, 2021-08-05:
Also notice that the sampling process is on the frame-level, which means we will keep 20% frames per sequence rather than 20% sequence of the whole training set.

Ultra11nstinct, 2021-08-10:
@jihanyang thanks for your reply and the contribution to this repo.
I have figured it out.
Wish every success in your work.

closed 2022-03-08 by github-actions[bot]