open-mmlab/OpenPCDet #607: Question about use_ball_constraint in PointHeadTemplate

gardenbaby, 2021-08-04:
I noticed that in `PointHeadTemplate`, if `use_ball_constraint` is set as `True`, it will add up the half of the box heights to calculate the ground truth box centers. But I thought the `gt_boxes` here are already transformed with true centers `gt_boxes[:,:3]`. 

    if set_ignore_flag:
         ... ...
    elif use_ball_constraint:
        box_centers = gt_boxes[k][box_idxs_of_pts][:, 0:3].clone()
        box_centers[:, 2] += gt_boxes[k][box_idxs_of_pts][:, 5] / 2              # here
        ball_flag = ((box_centers - points_single).norm(dim=1) < central_radius)
        fg_flag = box_fg_flag & ball_flag`

Did I get the wrong idea? Why add up the half of the heights?

wangguojun2018, 2022-02-09:
I have same question, I think it is not right for adding half box height to calculate groundtruth box centers

closed 2022-03-26 by github-actions[bot]

zhangtingyu11, 2022-10-28:
I have same question.