open-mmlab/OpenPCDet #611: Different AP@11 with released CaDDN model

djiajunustc, 2021-08-05:
Hi @codyreading ,

Thanks for your great efforts to propose CaDDN.
When I tried it in OpenPCDet with the docker image djiajun1206/pcdet:pytorch1.6, I find there are trivial differences in the predicted boxes. However, these trivial differences have caused obvious different AP@11.

Car AP@0.70, 0.70, 0.70:
bbox AP:89.9510, 80.1012, 78.7330
bev  AP:33.8986, 24.5491, 22.8564
3d   AP:26.0090, 19.7041, 16.7136
aos  AP:89.07, 78.96, 76.99
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:95.1532, 82.6433, 77.4363
bev  AP:31.7923, 21.6192, 19.4751
3d   AP:23.9281, 16.1892, 13.7052
aos  AP:94.11, 81.31, 75.67
Car AP@0.70, 0.50, 0.50:
bbox AP:89.9510, 80.1012, 78.7330
bev  AP:62.1695, 45.9791, 44.6544
3d   AP:57.7076, 43.3775, 37.5301
aos  AP:89.07, 78.96, 76.99
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:95.1532, 82.6433, 77.4363
bev  AP:62.5442, 46.1889, 42.2766
3d   AP:57.0426, 40.7876, 36.9711
aos  AP:94.11, 81.31, 75.67
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:48.5177, 40.6058, 37.2781
bev  AP:17.2814, 15.1830, 12.5905
3d   AP:15.2303, 12.9961, 11.4000
aos  AP:35.86, 30.07, 27.34
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:48.3889, 39.9524, 33.5314
bev  AP:13.8607, 10.2195, 8.0701
3d   AP:11.7012, 8.3523, 6.7415
aos  AP:33.36, 27.11, 22.48
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:48.5177, 40.6058, 37.2781
bev  AP:35.6592, 29.1780, 22.9294
3d   AP:33.7508, 28.0198, 22.5946
aos  AP:35.86, 30.07, 27.34
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:48.3889, 39.9524, 33.5314
bev  AP:32.5702, 25.6541, 20.3946
3d   AP:30.5830, 23.9918, 19.6985
aos  AP:33.36, 27.11, 22.48
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:35.6137, 24.4959, 23.4403
bev  AP:11.0340, 9.4251, 9.0909
3d   AP:10.7546, 9.0909, 9.0909
aos  AP:28.35, 20.37, 19.59
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:32.6001, 21.0963, 19.6310
bev  AP:2.8592, 1.6498, 1.2249
3d   AP:2.6598, 1.1575, 1.1245
aos  AP:24.03, 15.24, 14.31
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:35.6137, 24.4959, 23.4403
bev  AP:17.1759, 12.1026, 11.9617
3d   AP:16.9189, 12.0482, 11.9467
aos  AP:28.35, 20.37, 19.59
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:32.6001, 21.0963, 19.6310
bev  AP:12.5704, 6.3847, 6.0831
3d   AP:11.5313, 6.0065, 5.5909
aos  AP:24.03, 15.24, 14.31

I have ruled out many possible reasons that may account for these differences, but I failed to find out what it should be.
I think it may be due to the version of kornia.

Could you help me to resolve this problem?
Please try my docker image: djiajun1206/pcdet:pytorch1.6.
Thanks a lot.

Best Regards,
Jiajun Deng

codyreading, 2021-08-05:
Hi Jiajun,

We don't use AP@R11 for issues described [here](https://github.com/open-mmlab/OpenPCDet/issues/581). The results here look close enough to the results I have reported on R40, so I don't believe this is an issue.

djiajunustc, 2021-08-05:
Hi @codyreading,

Thanks for your reply. But I haven't trained the model again. Instead, I used the released parameters.

codyreading, 2021-08-05:
Ah okay, you can try the exact versions I use which are listed [here](https://github.com/TRAILab/CaDDN), and a docker file is even included there if you would like to use that. Additionally, I've also observed slightly different results when running the same code on different hardware, so that might be the cause as well.

djiajunustc, 2021-08-05:
I have also used your docker image and your released code. I get the same result as I reported.
Besides, I have tried it on different GPU devices, including DGX (v100), Titan V, RTX 2080 Ti.

djiajunustc, 2021-08-05:
That's so confusing.
I have loaded the result.pkl from mine and someone who can exactly reproduce your performance.
The differences are as follows:
[image]

I believe these differences are negligible.

codyreading, 2021-08-05:
This sounds like an issue in evaluation, then. I ran my inference with the provided weights and received the following results:
```
2021-08-05 10:16:48,669   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:89.9449, 80.0868, 78.7468
bev  AP:34.8573, 25.5907, 24.0973
3d   AP:27.7777, 21.3760, 18.6217
aos  AP:89.06, 78.95, 77.00
```

djiajunustc, 2021-08-05:
I think so. Please refer to the result I reported. The APs of AOS are almost the same.

Car AP@0.70, 0.70, 0.70:
bbox AP:89.9510, 80.1012, 78.7330
bev AP:33.8986, 24.5491, 22.8564
3d AP:26.0090, 19.7041, 16.7136
aos AP:89.07, 78.96, 76.99
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:95.1532, 82.6433, 77.4363
bev AP:31.7923, 21.6192, 19.4751
3d AP:23.9281, 16.1892, 13.7052
aos AP:94.11, 81.31, 75.67

closed 2021-08-06 by djiajunustc