open-mmlab/OpenPCDet #613: About the range of prediction's confidence

deepmeng, 2021-08-12:
What is the range of detection scores predicted by pointRCNN and PVRCNN? I noticed some work (https://github.com/xinshuoweng/AB3DMOT) used the detection results of pointRCNN and PVRCNN. But I feel the confidence values are strange, e.g., the confidence is negative or much larger than 1. In my impression, the confidence values in many methods are in a range of 0~1. How can I map these weird values to 0~1? @djiajunustc @sshaoshuai

djiajunustc, 2021-08-13:
Hi @deepmeng ,

Have you fed the output of the FC layer into the Sigmoid function?
The output range of the Sigmoid function should be 0~1.

deepmeng, 2021-08-14:
> Hi @deepmeng ,
> 
> Have you fed the output of the FC layer into the Sigmoid function?
> The output range of the Sigmoid function should be 0~1.

I have found the reason. Many thanks for your kind reply!

closed 2021-08-14 by deepmeng

amsvenkat, 2023-06-08:
HI @deepmeng 
What was the reason? Could you please let me know?

deepmeng, 2023-06-08:
> Hi @deepmeng ,
> 
> Have you fed the output of the FC layer into the Sigmoid function? The output range of the Sigmoid function should be 0~1.

@amsvenkat 
You can try this solution.