open-mmlab/OpenPCDet #623: CUDA kernel failed : no kernel image is available for execution on the device

ZJDWKTX, 2021-09-03:
when i run demo.py it back :

/usr/local/lib/python3.8/dist-packages/traits/etsconfig/etsconfig.py:412: UserWarning: Environment variable "HOME" not set, setting home directory to /tmp
  warn(
2021-09-03 05:01:48,065   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2021-09-03 05:01:48,262   INFO  Total number of samples:        1
2021-09-03 05:01:53,344   INFO  ==> Loading parameters from checkpoint pv_rcnn_8369.pth to CPU
2021-09-03 05:01:54,191   INFO  ==> Done (loaded 367/367)
2021-09-03 05:01:54,256   INFO  Visualized sample index:        1
CUDA kernel failed : no kernel image is available for execution on the devic

i have install 
cuda-11.1 torch1.9.0
i run torch.cuda.is_available() it back true
my gpu is quadro RTX 8000

why?

MartinHahner, 2021-09-03:
It looks like this type of GPU is unfortunately not supported (by some python package - maybe [spconv](https://github.com/traveller59/spconv)).

ZJDWKTX, 2021-09-06:
> It looks like this type of GPU is unfortunately not supported (by some python package - maybe [spconv](https://github.com/traveller59/spconv)).

thank you ,maybe i need run another computer!

Thomas-Kb, 2021-09-15:
> > It looks like this type of GPU is unfortunately not supported (by some python package - maybe [spconv](https://github.com/traveller59/spconv)).
> 
> thank you ,maybe i need run another computer!

You could try to use the spconv docker [image](https://hub.docker.com/r/scrin/dev-spconv) and continue as described in [this](https://yunshuang.medium.com/setting-up-running-environment-for-the-open-source-3d-object-detection-project-openpcdet-using-7427c0091edd) medium article. I managed to get OpenPCDet running with both a Maxwell and a Pascal GPU-architecture. I did not try with your Turing GPU-architecture though...

closed 2022-03-07 by github-actions[bot]