open-mmlab/OpenPCDet #626: Version of waymo-open-dataset tools

tdzdog, 2021-09-09:
I used to use tf-2.4.0 and the corresponding waymo-open-dataset tools. However, I found the trained model get very low mAP (10% on Vehicle). I notice that it is recommended to modify the version in (https://github.com/open-mmlab/OpenPCDet/issues/370). I am confused why the version has so much influence on the model performance. Is there any efficient methods to check the processed data? It is time-consuming to train on waymo.

jihanyang, 2021-09-10:
You can just visualize the point cloud with gts.

dkoguciuk, 2021-09-20:
@tdzdog , the crucial part here is [THIS](https://github.com/open-mmlab/OpenPCDet/issues/370#issuecomment-736683423) comment. If you have different numbers you have to use a different tf / waymo libs version. This is what worked for me:

```
tensorflow-cpu                           2.1.0
tensorflow-gpu                           2.1.0
waymo-open-dataset-tf-2-1-0              1.2.0
```

closed 2022-03-07 by github-actions[bot]