open-mmlab/OpenPCDet #627: Traning time on Waymo

tdzdog, 2021-09-12:
I am traning second on Waymo, with 5 3090 RTX and 6 sample on each gpu. However, I find it requires 20+ hours to train even with only 20% training set. Is such training speed normal? Or something is wrong?

MartinHahner, 2021-09-13:
I think the time estimate you state sounds reasonable. The Waymo dataset is huge.

tdzdog, 2021-09-22:
> > I am traning second on Waymo, with 5 3090 RTX and 6 sample on each gpu. However, I find it requires 20+ hours to train even with only 20% training set. Is such training speed normal? Or something is wrong?
> 
> How did you config the parameters of 'pin_memory' and 'num_workers' for your training in waymo dataset?

I use the default settings with 'pin_memory=True' and 'num_workers=8'

deepmeng, 2021-09-22:
I think your configurations and the training speed are pretty good. These configurations make great use of the CPU, memory and GPUs. Besides the model and some intermediate processes, the amount of dataset is definitely a major factor.

mqm-slice, 2021-12-17:
> I am traning second on Waymo, with 5 3090 RTX and 6 sample on each gpu. However, I find it requires 20+ hours to train even with only 20% training set. Is such training speed normal? Or something is wrong?

I have some problems when using the new 3090 server. Do you have docker images for 3090

roylu307, 2022-01-21:
Hi @MartinHahner  @tdzdog @jihanyang,
How did you get 20% of the training set? is it a default setting?

I noticed the two places to specify the sample interval:
https://github.com/open-mmlab/OpenPCDet/blob/8e055f02df245b1f0675e874792dd708fb84c04b/pcdet/datasets/waymo/waymo_dataset.py#L327 and
https://github.com/open-mmlab/OpenPCDet/blob/8e055f02df245b1f0675e874792dd708fb84c04b/tools/cfgs/dataset_configs/waymo_dataset.yaml#L13

Which takes the priority? I also noticed the generated processed data has a tag 'sampled_1', does it mean it is the full training set? If so, do I need to regenerate the data for the 20% set or is there a way to select 20% from the processed data? Thanks in advance

roylu307, 2022-01-21:
> How did you get 20% of the training set? is it a default setting?
>
> I noticed the two places to specify the sample interval: [...]

The first setting controls the data preprocessing, the second setting in the yaml file controls the sampling for the actual training? 
If I set the sample interval in the yaml file to 1, it trains on all processed data if I am guessing correctly.

jihanyang, 2022-01-21:
Yes.

closed 2022-01-21 by jihanyang