open-mmlab/OpenPCDet #629: How to train on custom dataset that has point cloud only but not image and calibration matrix ?

Fungungun, 2021-09-13:


jihanyang, 2021-09-14:
Many issues like 'custom datasets' have discussed this problem, so please refer to those first.

Fungungun, 2021-09-14:
> Many issues like 'custom datasets' have discussed this problem, so please refer to those first.

Hi jihan,

Thanks for your prompt reply. i have referred to all of them but still can't figure out where the problem. I have converted my data to the exact KITTI format and also create all the necessary dataset, config files etc . However i always got 0 AP during training. 

Is there any useful insights you may have ?

Thanks

jihanyang, 2021-09-14:
You can first visualize your dataset.

Fungungun, 2021-09-14:
Hi jihan,

Thanks. I have visualized the point clouds with the corresponding gt boxes and everything looks fine to me. 

When we convert the data to KITTI format, is the location supposed to be the bottom center or box center?

thanks

Thomas-Kb, 2021-09-15:
When you generate your dataset in KITTI format with 
`python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml`
as mentioned in ["Getting started"](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md), take a look at the generated _.bin_ files. If they dont look like your cropped ground truth, something is wrong with the labels. 

As mentioned in the [kitti paper](http://www.cvlibs.net/publications/Geiger2013IJRR.pdf) fig. 7, the x and y coordinates are in the middle of the object while the z coordinate is at the bottom. Keep in mind to rotate your labels and/or your pointcloud such that it corresponds to the kitti coordinate system. If you visualize your _gt_database_ and the _.bin_ files correspond to your objects, everything should be correct however.

YongYongLi123, 2021-10-16:
> When you generate your dataset in KITTI format with `python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml` as mentioned in ["Getting [...]

Hello, I visualized it in your way *. bin file in the database folder is found to be consistent with the label. The center point is in the center of the box, but the training result is still 0. The same dataset is visualized in mmdetection3d_ The. Bin in database is also found to be correct, and the training result evaluation in mmdetection3d is normal. I don't know why?

curiousboy20, 2021-11-16:
> What do you mean with "the training result is still 0"? Are you not able to train anything or is your trained model bad?

Hi, In my case the training was run well but it seems not convergence. So, when evaluating the dataset either with kitti pretrained model or custom trained model, the result is 0 mAP. This is the detail of my case: https://github.com/open-mmlab/OpenPCDet/issues/674

closed 2022-03-07 by github-actions[bot]

brillint, 2022-09-29:
kitti_dataset里的getitem里面可以看出，calib和image的作用就是求出lidar的label，如果你有lidar的label，直接替换就可以了

Petros626, 2024-10-01:
> Hello, I visualized it in your way *. bin file in the database folder is found to be consistent with the label. The center point is in the center of the box, but the training result is still 0. The [...]

Hey, when visualizing the files from the folder gt_database, where are the corresponding labels for it? Are the gt_database samples only loaded into the currently loaded sample during training or can you also access the data this way?