open-mmlab/OpenPCDet #633: Question about gt_sampling

darrenjkt, 2021-09-17:
Hi, 

My understanding of gt_sampling is that it addresses the class imbalance by over-sampling some of the less frequent classes. If this understanding is correct, why do models like voxel_rcnn_car use gt_sampling when it is only training on the car? 

Since it's only focusing on the "car" class, does it have any noticeable improvement when using gt_sampling vs not using gt_sampling?

dkoguciuk, 2021-09-17:
Hi @darrenjkt , 

I believe it's not a matter of fighting class imbalance, but foreground/background points imbalance. It was introduced in [ SECOND: Sparsely Embedded Convolutional Detection ](https://www.mdpi.com/1424-8220/18/10/3337) object detector:

> The major problem we encountered during training was the existence of too few ground truths,
> which significantly limited the convergence speed and final performance of the network. To solve this
> problem, we introduced a data augmentation approach. First, we generated a database containing
> the labels of all ground truths and their associated point cloud data (points inside the 3D bounding
> boxes of the ground truths) from the training dataset. Then, during training, we randomly selected
> several ground truths from this database and introduced them into the current training point cloud
> via concatenation. Using this approach, we could greatly increase the number of ground truths per
> point cloud and simulate objects existing in different environments. To avoid physically impossible
> outcomes, we performed a collision test after sampling the ground truths and removed any sampled
> objects that collided with other objects.

In fact, this is really powerful data augmentation technique, and you can see how much improvement it gives in @MartinHahner 's paper: [Quantifying Data Augmentation for LiDAR based 3D Object Detection](https://arxiv.org/abs/2004.01643)

darrenjkt, 2021-09-17:
I see that makes sense. Thanks for clarifying this gap in my knowledge and for showing me where to find more information.

closed 2021-09-17 by darrenjkt