open-mmlab/OpenPCDet #635: About argoverse dataset config

Shuntw6096, 2021-09-22:
Hi, I trained a point rcnn with kitti, and I want test on argoverse,  but get low AP on argoverse,
[image]
I translate argoverse dataset to kitti format using [Train in Germany, Test in The USA: Making 3D Object Detectors Generalize](https://github.com/cxy1997/3D_adapt_auto_driving)'s code, it seems working, and check kitti format with [kitti_object_vis](https://github.com/kuixu/kitti_object_vis).
Argoverse:
[image]
[image]
I think that some parameters settings are wrong because I use kitti's dataset setting, and I don't know how to set argoverse's `POINT_CLOUD_RANGE`, how to get  `POINT_CLOUD_RANGE` parameters.
argoverse datset setting below:
```
DATASET: 'ArgoDataset'
DATA_PATH: '../data/argo'

POINT_CLOUD_RANGE: [0, -40, -3, 200, 40, 1]

DATA_SPLIT: {
    'train': train,
    'test': val
}

INFO_PATH: {
    'train': [kitti_infos_train.pkl],
    'test': [kitti_infos_val.pkl],
}

GET_ITEM_LIST: ["points"]
FOV_POINTS_ONLY: True

DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
        - NAME: gt_sampling
          USE_ROAD_PLANE: False
          DB_INFO_PATH:
              - kitti_dbinfos_train.pkl
          PREPARE: {
             filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5'],
             filter_by_difficulty: [-1],
          }

          SAMPLE_GROUPS: ['Car:20','Pedestrian:15', 'Cyclist:15']
          NUM_POINT_FEATURES: 4
          DATABASE_WITH_FAKELIDAR: False
          REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
          LIMIT_WHOLE_SCENE: True

        - NAME: random_world_flip
          ALONG_AXIS_LIST: ['x']

        - NAME: random_world_rotation
          WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

        - NAME: random_world_scaling
          WORLD_SCALE_RANGE: [0.95, 1.05]

POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}

DATA_PROCESSOR:
    - NAME: mask_points_and_boxes_outside_range
      REMOVE_OUTSIDE_BOXES: True

    - NAME: shuffle_points
      SHUFFLE_ENABLED: {
        'train': True,
        'test': False
      }

    - NAME: transform_points_to_voxels
      VOXEL_SIZE: [0.05, 0.05, 0.1]
      MAX_POINTS_PER_VOXEL: 5
      MAX_NUMBER_OF_VOXELS: {
        'train': 16000,
        'test': 40000
      }
```
I use [model setting](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/kitti_models/pointrcnn.yaml) but change ```CLASS_NAMES``` to ```['Car']```
Thanks.

jihanyang, 2021-09-23:
Hello, please refer #253 to set the point cloud range and voxel size. You need to make point cloud arange to cover ring view point cloud.

Shuntw6096, 2021-09-27:
Hi, @jihanyang, I iterate ```velodyne``` directory to get minimum and maximum value along x, y, and z axis (.bin file format: x, y, z, reflectance, row by row), and I get (min x, min y, min z, max x, max  y, max z), 
``` yaml
[-99.17721558 -11.09814548  -0.41421485 220.19508362 217.98701477  94.82263184]
 ```
and change the values with [https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-679190936](https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-679190936), get
``` yaml
[-99.25, -11.12, -0.43, 220.27, 218.0, 79.57]
```
but I ran into some trouble when running 
``` bash
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos $YAML_PATH
```
the processes directly had been killed, I have no idea.
``` bash
root@435b832144c1:/home/OpenPCDet# python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/argo_dataset_kitti.yaml 
/usr/lib/python3.8/runpy.py:127: RuntimeWarning: 'pcdet.datasets.kitti.kitti_dataset' found in sys.modules after import of package 'pcdet.datasets.kitti', but prior to execution of 'pcdet.datasets.kitti.kitti_dataset'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
/home/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py:481: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  dataset_cfg = EasyDict(yaml.load(open(sys.argv[2])))
path--------------------------------------------- /home/OpenPCDet/data/argo
---------------Start to generate data infos---------------
Killed
root@435b832144c1:/home/OpenPCDet# 
```
but I multiple 0.8 to the values and run ```python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos $YAML_PATH``` and it seems working.
``` yaml
[-79.4, -8.896, -0.344, 176.216, 174.4, 63.656]
```
and I change [pointrcnn.yaml](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/kitti_models/pointrcnn.yaml)'s ```DEPTH_NORMALIZER``` to ```176.0```.
Finally, I test on argoverse, and get low AP on argoverse, what settings do I need to adjust?
Thanks.

closed 2022-03-07 by github-actions[bot]

BBBeatrice, 2023-08-26:
@Shuntw6096 Hi, I am now working on a model based on openPCDet using argoverse dataset as input, and I encounter the same problem as you when I test on the model the process is killed. I have no idea what is wrong, have you solved your problem? Could you please share the solution please? Thanks.

yinheyanxian, 2023-12-16:
> problem as you when I test on the model the process is killed. I have no idea what is wrong, have you solved your problem? Could

Hello, I have the same problem. Have you resolved it?