open-mmlab/OpenPCDet #639: Voxel R-CNN on nuScenes dataset

deepmeng, 2021-09-27:
@djiajunustc Hi Jiajun, thanks for your excellent work of Voxel R-CNN. 
I have tried your Voxel R-CNN model on the datasets of KITTI and Waymo for multi-category detection. And the model performed very well. However, when I tried on the nuScenes dataset, I met the issue like this:
` File "/pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 148, in sample_rois_for_rcnn
    batch_gt_of_rois[index] = cur_gt[gt_assignment[sampled_inds]]
RuntimeError: The expanded size of the tensor (8) must match the existing size (10) at non-singleton dimension 1.  Target sizes: [128, 8].  Tensor sizes: [128, 10] `.

I have tried to analyze and debug the code and model, so far there is no answer. I cannot figure out why the ground-truth's code shape became to 10. Can you provide me with any enlightenment?

CSautier, 2021-09-27:
Hi,
I've had the same issue. It has to do with NuScenes also having the information of velocity on x and y (hence the 2 additional dimensions).
A quick (albeit ugly) fix is to drop the dimensions corresponding to the velocity. You can do that in the class NuScenesDataset, (line 33), you will also have to do a similar operation for the augmentation in database_sampler.py (line 179).

deepmeng, 2021-09-27:
> Hi,
> I've had the same issue. It has to do with NuScenes also having the information of velocity on x and y (hence the 2 additional dimensions).
> A quick (albeit ugly) fix is to drop the dimensions [...]

Hi, man, thanks for your reply. 

I noticed the code of
` if not self.dataset_cfg.PRED_VELOCITY and 'gt_boxes' in data_dict:
        Data_dict [' gt_boxes] = data_dict [' gt_boxes] [: [0, 1, 2, 3, 4, 5, 6, 1]] ` in the dattabase_sample.py file.
Before you answer my question, I tried adding a key-value pair 'PRED_VELOCITY: False' in the 'DATA_CONFIG' field in the model definition file (.yaml). This action blocks the two extra speed dimensions you mentioned and also avoids unnecessary modifications to the code. I am training and the training is normal now. I have not evaluated the performance of the model as the training has not ended now. 

Thank you again!

closed 2021-09-27 by deepmeng

yinggo, 2021-10-03:
Hi @deepmeng , recently, I'd like to use NuScenes datasets, but I am not sure about the performance of Voxel R-CNN. How are your results of Voxel R-CNN on NuScenes?

deepmeng, 2021-10-03:
> Hi @deepmeng , recently, I'd like to use NuScenes datasets, but I am not sure about the performance of Voxel R-CNN. How are your results of Voxel R-CNN on NuScenes?

These are my evaluation results of Voxel R-CNN on the nuScenes dataset.
[image]

yinggo, 2021-10-03:
@deepmeng Thanks!!

Eaton2022, 2023-10-06:
> > Hi @deepmeng , recently, I'd like to use NuScenes datasets, but I am not sure about the performance of Voxel R-CNN. How are your results of Voxel R-CNN on NuScenes?
> 
> These are my evaluation results of Voxel R-CNN on the nuScenes dataset. [image]

请问这是练了多少epoch的结果？这结果表明Voxel-RCNN在nuScenes上表现不佳么