open-mmlab/OpenPCDet #646: AP is 0 when model is trained for more then one anchor box height (anchor_bottom_heights)

crypto-g245, 2021-10-05:
Hello, 

I'm training PVRCNN model with a custom dateset to detect the objects above the ground. These objects are at different heights from the grounds. 
Is it possible to use multiple anchors to detect these objects accurately. 
I used two values for 'anchor_bottom_heights' parameter in model config file. but the loss was stagnant after few epochs and AP is 0.  

Is it anyways possible to use multiple anchors? or am I missing some additional changes in model config?

YongYongLi123, 2021-10-16:
The AP I got from training pointpillars with a user-defined dataset is also 0. Have you solved this problem

sshaoshuai, 2022-02-21:
We never tried to use multiple heights for anchor-based 3D detections, so it may needs to spend lots of time to support this feature (or maybe we just need some small modifications). 

I think this feature is not important in autonomous driving scenarios.
For 3D detection in other scenarios, generally we will use some PointHead-based 3D detector (like PointRCNN) to deal with this problem for 3D detection in the whole 3D space instead of bird's eye view. 

So currently there is no plan to further investigate it.

Welcom to submit a PR for this if you find some small bugs about this problem.

crypto-g245, 2022-02-21:
Hi @sshaoshuai 

Thanks for the quick response and hint !!! 
I will test it with PointRCNN again and see how it works.

closed 2022-04-08 by github-actions[bot]