open-mmlab/OpenPCDet #660: i got error when i use"python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml"

Liyan943, 2021-11-05:
anyone know the reason? thanks a lot.

Traceback (most recent call last):
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/lijunda/桌面/env_learing/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 482, in <module>
    save_path=ROOT_DIR / 'data' / 'kitti'
  File "/home/lijunda/桌面/env_learing/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 448, in create_kitti_infos
    kitti_infos_val = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
  File "/home/lijunda/桌面/env_learing/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 222, in get_infos
    return list(infos)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/concurrent/futures/_base.py", line 598, in result_iterator
    yield fs.pop().result()
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/concurrent/futures/_base.py", line 428, in result
    return self.__get_result()
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/lijunda/桌面/env_learing/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 159, in process_single_scene
    image_info = {'image_idx': sample_idx, 'image_shape': self.get_image_shape(sample_idx)}
  File "/home/lijunda/桌面/env_learing/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 86, in get_image_shape
    return np.array(io.imread(img_file).shape[:2], dtype=np.int32)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/skimage/io/_io.py", line 48, in imread
    img = call_plugin('imread', fname, plugin=plugin, **plugin_args)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/skimage/io/manage_plugins.py", line 207, in call_plugin
    return func(*args, **kwargs)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/skimage/io/_plugins/imageio_plugin.py", line 10, in imread
    return np.asarray(imageio_imread(*args, **kwargs))
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/imageio/core/functions.py", line 159, in imread
    return file.read(index=0, **kwargs)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/imageio/core/imopen.py", line 263, in read
    reader = self.legacy_get_reader(**kwargs)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/imageio/core/imopen.py", line 242, in legacy_get_reader
    return self._plugin.get_reader(req)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/imageio/core/format.py", line 163, in get_reader
    return self.Reader(self, request)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/imageio/core/format.py", line 213, in __init__
    self._open(**self.request.kwargs.copy())
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/imageio/plugins/pillow_legacy.py", line 391, in _open
    return PillowFormat.Reader._open(self, pilmode=pilmode, as_gray=as_gray)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/imageio/plugins/pillow_legacy.py", line 297, in _open
    pil_try_read(self._im)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/imageio/plugins/pillow_legacy.py", line 651, in pil_try_read
    im.getdata()[0]
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/PIL/Image.py", line 1319, in getdata
    self.load()
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/PIL/ImageFile.py", line 237, in load
    s = read(self.decodermaxblock)
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/PIL/PngImagePlugin.py", line 896, in load_read
    cid, pos, length = self.png.read()
  File "/home/lijunda/anaconda3/envs/openpcdet/lib/python3.7/site-packages/PIL/PngImagePlugin.py", line 166, in read
    raise SyntaxError(f"broken PNG file (chunk {repr(cid)})")
SyntaxError: broken PNG file (chunk b'\x00\x00\x00\x00')

curiousboy20, 2021-11-17:
I think one of the image files in your dataset error. If you use kitti dataset, have you tried to redownload the image files?

closed 2022-03-07 by github-actions[bot]

anton-starodub, 2022-05-04:
Please, let me know if you have found a solution to this one.

Even skipping this particular image would work great for me.