open-mmlab/OpenPCDet #661: substantial accuracy difference on different classes

ChunmianLin, 2021-11-07:
Hi, authors. I reproduced your experimental results using the pretrained model and do not modify any training settings. But huge accuracy difference on various classes compared with that in the paper, i.e. 79.10%(ours) vs. 81.43%(paper) in Car moderate level, 35.20%(ours) vs. 43.29%(paper) in Pedestrian moderate level, and 57.72%(ours) vs. 63.71%(paper) in Cyclist moderate level. 
What is the problem that might cause these performance gaps? Any advice for improving the results?
Thanks in advance!

xiazhongyv, 2021-12-21:
Maybe you should use the test code to test for every epoch and see the tensorboard curve. Because there always be sharp change of val performance on KITTI.
You can see the relevant discussion at #189 #314

ChunmianLin, 2022-01-21:
@xiazhongyv Thanks. The performance gap on 'pedestrian' and 'cyclist' are too substantial to explain. I have visualized the tensorboard curve during the training and test phase, but there is no abnormality. 
Do you have any better advice to enhance the accuracy performance, particularly in these two categories? Thanks in advance!

closed 2022-03-07 by github-actions[bot]