open-mmlab/OpenPCDet #662: Got error : TypeError: load() missing 1 required positional argument: 'Loader' when create kitti infos

curiousboy20, 2021-11-07:
Hi, I just installed this repo and the demo was run well. However, when I try to train kitti data and get infos file with command:
`python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml tools/cfgs/dataset_configs/kitti_dataset.yaml
`
there was error with detail:
`/home/yonas/anaconda3/envs/spconv12/lib/python3.8/runpy.py:127: RuntimeWarning: 'pcdet.datasets.kitti.kitti_dataset' found in sys.modules after import of package 'pcdet.datasets.kitti', but prior to execution of 'pcdet.datasets.kitti.kitti_dataset'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
Traceback (most recent call last):
  File "/home/yonas/anaconda3/envs/spconv12/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/yonas/anaconda3/envs/spconv12/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/yonas/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 476, in <module>
    dataset_cfg = EasyDict(yaml.load(open(sys.argv[2])))
TypeError: load() missing 1 required positional argument: 'Loader'
`

Please help

curiousboy20, 2021-11-07:
Solved by changing yaml.load to yaml.safe_load according to :
https://stackoverflow.com/questions/69564817/typeerror-load-missing-1-required-positional-argument-loader-in-google-col
and 
https://stackoverflow.com/questions/1773805/how-can-i-parse-a-yaml-file-in-python/1774043

closed 2021-11-07 by curiousboy20

JiangZhenW, 2022-05-17:
you can solve this by adding "Loader = yaml.FullLoader" after "(yaml.load(open(sys.argv[2])", just become "(yaml.load(open(sys.argv[2]), Loader = yaml.FullLoader"