open-mmlab/OpenPCDet #664: Bug - gt_annos might be removed after data preprocessing

KickCellarDoor, 2021-11-08:
The point cloud range filter might remove some existing object annotations. And if there is no gt_annos inside of the whole batch, an error will be triggered for the Kitti dataset.

Fix: resample if there is no gt_annos after data preprocessing.
```
        data_dict = self.prepare_data(data_dict=input_dict)

        data_dict['image_shape'] = img_shape

        if ('annos' not in info) or (len(data_dict['gt_boxes']) > 0) \
                or (self.split == 'val') or (self.split == 'test'):
            return data_dict
        else:
            resample_idx = random.choice(list(range(len(self))))
            print(f"There is no gt_boxes, resample idx {resample_idx} from dataset!")
            return self.__getitem__(resample_idx)
```

jihanyang, 2021-11-14:
We already implemented this in prepare_data.

closed 2021-11-14 by jihanyang

KickCellarDoor, 2021-11-14:
Sorry, I am using an old version of OpenPCDet. Should check your latest code first.