open-mmlab/OpenPCDet #665: Dataset size sensitivity analysis with KITTI and pv_rcnn

JenWei0312, 2021-11-09:
Hi Authors or anyone who can help,
I performed multiple trainings for pv_rcnn with different sizes of KITTI data: 1K, 2K, 4K, and full data. Train/val splits are based on ImageSets protocol and both data and model configs are as the default, and I'm augmenting data in all runs.

I noticed that :
1. The overall train loss was the lowest for 4K data (not the full set). 
2. The full-data trained model only performs(R40) the best for car and cyclist in most cases; it performs consistently worse than both 4K and 2K data train models for the pedestrian class.

I am wondering if these observation are expected. If not, did I do anything wrong. If the observations are expected, what could be the reason(s)?

Thank you.

MartinHahner, 2021-11-25:
@JenWei0312 
Your observations are not expected, but they are also not completely unreasonable. It could e.g. be that the pedestrian class is underrepresented in the second half of the dataset or at least in the samples higher than 4k. That would explain why the training runs with the full data perform worse on pedestrians. But this is just a guess. It could also very be something else I don't think of right now.

JenWei0312, 2021-12-10:
I see, thank you for sharing your thoughts.

closed 2021-12-10 by JenWei0312