open-mmlab/OpenPCDet #667: conflict with spconv

who1129, 2021-11-10:
torch==1.7.0
cuda==11.0
install spconv `pip install spconv-cu111`
GPU: RTX3090

```
root@a32cb263f759:/home//OpenPCDet/tools# python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt ../pv_rcnn_8369.pth     --data_path /home//Dataset/KITTI/training/velodyne/000000.bin 
/usr/local/lib/python3.8/dist-packages/traits/etsconfig/etsconfig.py:409: UserWarning: Environment variable "HOME" not set, setting home directory to /tmp
  warn(
Traceback (most recent call last):
  File "demo.py", line 11, in <module>
    from pcdet.models import build_network, load_data_to_gpu
  File "/home//OpenPCDet/pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/home/OpenPCDet/pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 7, in <module>
    from .. import backbones_2d, backbones_3d, dense_heads, roi_heads
  File "/home/OpenPCDet/pcdet/models/backbones_3d/__init__.py", line 2, in <module>
    from .spconv_backbone import VoxelBackBone8x, VoxelResBackBone8x
  File "/home/OpenPCDet/pcdet/models/backbones_3d/spconv_backbone.py", line 29, in <module>
    class SparseBasicBlock(spconv.SparseModule):
AttributeError: module 'spconv' has no attribute 'SparseModule'
```

Then I found this method -> https://githubmemory.com/repo/traveller59/spconv/issues/324

modifying import  part of `spconv_backobne.py` and s`pconv_unet.py`

But still have a problem... from spconv.utils import VoxelGenerator

```
`root@a32cb263f759:/home/OpenPCDet/tools# python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt ../pv_rcnn_8369.pth     --data_path /home/ext/Dataset/KITTI/training/velodyne/000000.bin 
/usr/local/lib/python3.8/dist-packages/traits/etsconfig/etsconfig.py:409: UserWarning: Environment variable "HOME" not set, setting home directory to /tmp
  warn(
2021-11-10 03:03:32,241   INFO  -----------------Quick Demo of OpenPCDet-------------------------
Traceback (most recent call last):
  File "/home/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 50, in transform_points_to_voxels
    from spconv.utils import VoxelGeneratorV2 as VoxelGenerator
ImportError: cannot import name 'VoxelGeneratorV2' from 'spconv.utils' (/usr/local/lib/python3.8/dist-packages/spconv/utils/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "demo.py", line 103, in <module>
    main()
  File "demo.py", line 76, in main
    demo_dataset = DemoDataset(
  File "demo.py", line 26, in __init__
    super().__init__(
  File "/home/OpenPCDet/pcdet/datasets/dataset.py", line 33, in __init__
    self.data_processor = DataProcessor(
  File "/home/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 17, in __init__
    cur_processor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "/home/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 52, in transform_points_to_voxels
    from spconv.utils import VoxelGenerator
ImportError: cannot import name 'VoxelGenerator' from 'spconv.utils' (/usr/local/lib/python3.8/dist-packages/spconv/utils/__init__.py)`
```
Please help to use the OpenPCDet library in this environment!

mike1999888, 2021-11-21:
Me toooo.....The question  is that the version of spconv is 2.x but this program uses 1.0/1.2, I found the previous version but it can't pass by compiling.....

xiazhongyv, 2021-11-25:
#658 It's OK now.

closed 2021-12-01 by sshaoshuai

SaMeEr9597, 2022-03-12:
> torch==1.7.0 cuda==11.0 install spconv `pip install spconv-cu111` GPU: RTX3090
>
> ```
> root@a32cb263f759:/home//OpenPCDet/tools# python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt [...]

Hi, I am facing the same issue, what should we modify in the import part of these files, could you please explain in detail as I am new to this

SaMeEr9597, 2022-03-12:
> torch==1.7.0 cuda==11.0 install spconv `pip install spconv-cu111` GPU: RTX3090
>
> ```
> root@a32cb263f759:/home//OpenPCDet/tools# python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt [...]

When I am changing the spconv_backobne.py and spconv_unet.py according to your suggestion, I am getting this error: 
[image]