open-mmlab/OpenPCDet #670: Did you shuffle the dataset's original frame IDs?

deepmeng, 2021-11-13:
Hi @sshaoshuai and @djiajunustc,
Take the KITTI dataset for example, when you create the database for KITTI, did you shuffle the original frame IDs and generate new IDs? When debugging the code, I found that the frame_id and gt_boxes provided by data_dict cannot match the KITTI dataset in same frame id. If this is true, how can I find the original kitti frame_id based on the frame_id in data_dict?

deepmeng, 2021-11-13:
Hi @sshaoshuai, @djiajunustc and @MartinHahner,
I think the 'mismatch between the data_dict and origin KITTI dataset on the same frame_id' may results in data augmentation.

I have two new questions. 
(1) When training the model, if we set the DATA_AUGMENTOR in the model. yaml, I guess the DATA_AUGMENTOR in kitti_dataset.yaml will be overwritten in the code, right? 
(2) If there is no DATA_AUGMENTOR in the model.yaml, it will always generate new samples as the DATA_AUGMENTOR in kitti_dataset.yaml, right?

jihanyang, 2021-11-14:
We do not shuffle frame ids

deepmeng, 2021-11-14:
> We do not shuffle frame ids

Hi @jihanyang, thanks for your reply. As I mentioned, the data augmentation made me misunderstand the 'mismatch'. 

Furthermore, do you think the two new issues are right?

closed 2021-11-14 by deepmeng