open-mmlab/OpenPCDet #671: Questions about using your own data set visualization

ferrari700, 2021-11-14:
Hello, OpenPCDet
team! Thank you so much for your code! I used your voxel-rcnn pretained model to test my own data set, and adjusted the POINT_CLOUD_RANGE in kitti_dataset.yaml, but the point cloud image is incomplete and the coordinate system is still problematic. How can I solve it? Very grateful for your help！！
[image]

closed 2021-11-24 by ferrari700

DavidWZX, 2022-01-05:
Hello @ferrari700 ! Have you solved the problem? I also met lots of problem when I run demo and test file on my own dataset. Can you share some tricks, please? Thanks a lot!

MartinHahner, 2022-01-05:
Maybe you two can leverage my own [pointcloud_viewer.py](https://github.com/MartinHahner/LiDAR_fog_sim/blob/main/pointcloud_viewer.py).