open-mmlab/OpenPCDet #673: ValueError: Could not load ""  Reason: "unrecognized data stream contents when reading image file"

VE-yyq, 2021-11-15:
Thanks for your work!
When I run `python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml` to generate the data, I get the following error. Can you please tell me how to solve it?

`Traceback (most recent call last):
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/imageio/plugins/pillow.py", line 669, in pil_try_read
    im.getdata()[0]
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/PIL/Image.py", line 1282, in getdata
    self.load()
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/PIL/ImageFile.py", line 284, in load
    raise_oserror(err_code)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/PIL/ImageFile.py", line 67, in raise_oserror
python-BaseException
    raise OSError(message + " when reading image file")
OSError: unrecognized data stream contents when reading image file
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/concurrent/futures/_base.py", line 619, in result_iterator
    yield fs.pop().result()
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/concurrent/futures/_base.py", line 437, in result
    return self.__get_result()
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/adas/devdata4/yyq/code/vision3d/CaDDN-master/CaDDN-master/pcdet/datasets/kitti/kitti_dataset.py", line 184, in process_single_scene
    image_info = {'image_idx': sample_idx, 'image_shape': self.get_image_shape(sample_idx)}
  File "/home/adas/devdata4/yyq/code/vision3d/CaDDN-master/CaDDN-master/pcdet/datasets/kitti/kitti_dataset.py", line 108, in get_image_shape
    return np.array(io.imread(img_file).shape[:2], dtype=np.int32)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/skimage/io/_io.py", line 48, in imread
    img = call_plugin('imread', fname, plugin=plugin, **plugin_args)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/skimage/io/manage_plugins.py", line 207, in call_plugin
    return func(*args, **kwargs)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/skimage/io/_plugins/imageio_plugin.py", line 10, in imread
    return np.asarray(imageio_imread(*args, **kwargs))
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/imageio/core/functions.py", line 265, in imread
    reader = read(uri, format, "i", **kwargs)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/imageio/core/functions.py", line 186, in get_reader
    return format.get_reader(request)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/imageio/core/format.py", line 170, in get_reader
    return self.Reader(self, request)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/imageio/core/format.py", line 221, in __init__
    self._open(**self.request.kwargs.copy())
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/imageio/plugins/pillow.py", line 298, in _open
    return PillowFormat.Reader._open(self, pilmode=pilmode, as_gray=as_gray)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/imageio/plugins/pillow.py", line 135, in _open
    pil_try_read(self._im)
  File "/home/torch/anaconda3/envs/pcdet/lib/python3.8/site-packages/imageio/plugins/pillow.py", line 680, in pil_try_read
    raise ValueError(error_message)
ValueError: Could not load "" 
Reason: "unrecognized data stream contents when reading image file"
Please see documentation at: http://pillow.readthedocs.io/en/latest/installation.html#external-libraries
`

VE-yyq, 2021-11-16:
solved

closed 2021-11-16 by VE-yyq

zhen12079, 2022-03-10:
hi， i met the same issue. Can you give some advices? wechat: 18855486885

zhen12079, 2022-03-10:
> I encountered the same problem: OSError: unrecognized data stream contents when reading image file. How did you solve it?

solved?

Molly260, 2022-03-10:
yeah, I have solved this error. I converted the image to "RGB" by using Image.open(path).convert("RGB")

zhen12079, 2022-03-10:
can i chat with you by wechat?18855486885
i don`t know where file to  converted the image to "RGB" by using Image.open(path).convert("RGB")

zhen12079, 2022-03-10:
> i don`t know where file to converted the image to "RGB" by using Image.open(path).convert("RGB")

i don`t know where file to converted the image to "RGB" by using Image.open(path).convert("RGB")
can you give me some points

zhen12079, 2022-03-15:
> solved

hi.can you tell how to slove?

jkboyJohn, 2022-08-17:
> > solved
> 
> hi.can you tell how to slove?

hello, I also meet this bug, could you tell me how to solve it?

sayhiOK, 2023-04-22:
> solved
@VE-yyq How did you solve it?

sayhiOK, 2023-04-22:
> > 解决了
> 
> 嗨。你能告诉我怎么爱吗？

@zhen12079 Did you solve this problem?