open-mmlab/OpenPCDet #675: ptxas application ptx input, line 9; fatal   : Unsupported .version 7.3; current version is '7.2'

18303848764, 2021-11-17:
hello,I was testing test.py and then there are some bug ,I can't deal with them ,I want to be helped.

For more information about alternatives visit: ('https://numba.pydata.org/numba-doc/latest/cuda/overview.html', '#cudatoolkit-lookup')
  warnings.warn(errors.NumbaWarning(msg))
Traceback (most recent call last):
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/cudadrv/driver.py", line 2161, in add_ptx
    ptxbuf, len(ptx), namebuf, 0, None, None)
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/cudadrv/driver.py", line 300, in safe_cuda_api_call
    self._check_error(fname, retcode)
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/cudadrv/driver.py", line 335, in _check_error
    raise CudaAPIError(retcode, msg)
numba.cuda.cudadrv.driver.CudaAPIError: [222] Call to cuLinkAddData results in UNKNOWN_CUDA_ERROR

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 199, in <module>
    main()
  File "test.py", line 195, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 63, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/han/EDisk/OpenPCDet/tools/eval_utils/eval_utils.py", line 113, in eval_one_epoch
    output_path=final_output_dir
  File "/home/han/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 361, in evaluation
    from .kitti_object_eval_python import eval as kitti_eval
  File "/home/han/OpenPCDet/pcdet/datasets/kitti/kitti_object_eval_python/eval.py", line 6, in <module>
    from .rotate_iou import rotate_iou_gpu_eval
  File "/home/han/OpenPCDet/pcdet/datasets/kitti/kitti_object_eval_python/rotate_iou.py", line 263, in <module>
    def rotate_iou_kernel_eval(N, K, dev_boxes, dev_query_boxes, dev_iou, criterion=-1):
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/decorators.py", line 95, in kernel_jit
    return Dispatcher(func, [func_or_sig], targetoptions=targetoptions)
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/compiler.py", line 899, in __init__
    self.compile(sigs[0])
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/compiler.py", line 1102, in compile
    kernel.bind()
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/compiler.py", line 590, in bind
    self._func.get()
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/compiler.py", line 441, in get
    linker.add_ptx(ptx)
  File "/home/han/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/numba/cuda/cudadrv/driver.py", line 2163, in add_ptx
    raise LinkerError("%s\n%s" % (e, self.error_log))
numba.cuda.cudadrv.driver.LinkerError: [222] Call to cuLinkAddData results in UNKNOWN_CUDA_ERROR
ptxas application ptx input, line 9; fatal   : Unsupported .version 7.3; current version is '7.2'

closed 2022-03-07 by github-actions[bot]

saurtone, 2022-04-16:
我也遇到了同样的问题

20210726, 2022-05-05:
同样的问题

Eaphan, 2022-05-12:
Same question

saurtone, 2022-10-11:
邮件我已收到。辛苦了。

gmarkall, 2022-10-12:
This issue occurs when the toolkit version is newer than the CUDA version supported by the driver, because [Minor Version Compatibility](https://docs.nvidia.com/deploy/cuda-compatibility/index.html#minor-version-compatibility) is not yet supported by Numba. I have a Numba PR to add support for it: https://github.com/numba/numba/pull/8180

In the meantime, you can also work around the issue by installing the `cubinlinker` and [`ptxcompiler`](https://github.com/rapidsai/ptxcompiler) packages:

```
conda install rapidsai::cubinlinker conda-forge::ptxcompiler
```

and using the Numba patch before running any Numba CUDA functionality:

```python
from cublinlinker.patch import patch_numba_linker_if_needed
patch_numba_linker_if_needed()
```