open-mmlab/OpenPCDet #677: Why use multi head?

ahmedius2, 2021-11-17:
Hello,

I have two questions.
1- Most likely using multiple heads decreases the runtime speed compared to a single head. Why are the nuscenes models are designed with multi-head? Does it provide much better mAP compared to single?
2- AFAIK vision detection models such as YOLO do not use multiple heads (separated according to classes). Is using multi-head something specific for LiDAR? If yes, why?

Many thanks for this great repository.
Best regards,

zye1996, 2021-11-19:
> Hello,
>
> I have two questions. 1- Most likely using multiple heads decreases the runtime speed compared to a single head. Why are the nuscenes models are designed with multi-head? Does it provide [...]

1. Generally multi-head will boost mAP especially for nuscenes when points are scarce while it has more classes in many different scales.
2.  Multi-head slows down inference speed. I believe 2D object detection mostly skipped multi-head as they do not have an absolute anchor size for each class (since it depends on the camera-object distance) and each anchor can be any class. Say you have three classes, an anchor can be each of dog, bird, and cat. You do not want to make classification individually as you know a single anchor can only be one of those three classes. It is more straightforward in the case of lidar to decide whether an anchor matches a groundtruth of a specific class when we calculate the IoU between fixed anchor size and groundtruth.

ahmedius2, 2021-11-24:
Many thanks for the answer! Could you explain a little more by what you meant with the points being scarce? In which scenarios that would be the case? Also if possible, can you provide an example point cloud from the dataset where multi-head would be especially useful?

zye1996, 2021-11-24:
> Many thanks for the answer! Could you explain a little more by what you meant with the points being scarce? In which scenarios that would be the case? Also if possible, can you provide an example point cloud from the dataset where multi-head would be especially useful?

Point is scarce means there are only a few points for each object (especially small objects) and a total number of points is small compared to the entire space.

MartinHahner, 2021-11-25:
nuScenes example (~20k points)
[image]

KITTI example (~120k points)
[image]

ahmedius2, 2021-11-25:
Again, many thanks.

closed 2021-11-25 by ahmedius2