open-mmlab/OpenPCDet #678: Can I see AP during training?

kx-Z, 2021-11-19:


MartinHahner, 2021-11-25:
Not as of right now. Would be nice to have though.

kx-Z, 2021-11-26:
> Not as of right now. Would be nice to have though.
Oh, thank you. That's a pity.

sshaoshuai, 2021-12-01:
We do not support this since it will greatly slow down the training process.  

A potential strategy is to run a `test.py` with `--eval_all` simultaneously and you could see the AP curve from tensorboard.

kx-Z, 2021-12-01:
> We do not support this since it will greatly slow down the training process.
> 
> A potential strategy is to run a `test.py` with `--eval_all` simultaneously and you could see the AP curve from tensorboard.

Thank you，got it. That‘s different from using a small val set when i was training a 2D model.

MartinHahner, 2021-12-08:
> We do not support this since it will greatly slow down the training process.
> 
> A potential strategy is to run a `test.py` with `--eval_all` simultaneously and you could see the AP curve from tensorboard.

I forgot about that option. I think this approach is enough to track the process **while** training and no further coding is required.

closed 2021-12-10 by sshaoshuai