open-mmlab/OpenPCDet #679: About result.pkl generated by test.py

ObsisMc, 2021-11-22:
I trained kitti data and used test.py to test my model and it generate result.pkl. However, after comparing labels of kitti and data in result.pkl, I found that there are some differences.
1. dimension attribute of result.pkl is in order of length, height and width but labels of kitti is in order of height, width and length.
2. alpha and rotation_y attributes of result.pkl and labels of kitti have a difference of 2*pi.

Is my analysis right? If not, what should I do to adjust the result.

sshaoshuai, 2021-12-01:
The results of `result.pkl` are formatted with the unified coordinate of OpenPCDet, which is different with the KITTI submission format. 

Please look at here to see how to transform our unified coordinate to KITTI format (actually you just need to add `--save_to_file` command line argument when running `test.py`, to generate results with KITTI format): 
https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/kitti/kitti_dataset.py#L329-L349

hz3014, 2021-12-03:
> at here to see how to transform our unified coordinate to KITTI format

Hi, shaoshuai, I was wondering if there is code to convert result.pkl to .bin file for watmo server evaluation?

sshaoshuai, 2021-12-09:
Currently this part is not included in OpenPCDet. You may write it by yourself by referring to the submission format of official Waymo repo, and it should not be very difficult.

closed 2021-12-09 by sshaoshuai