open-mmlab/OpenPCDet #681: Error while installing PCDet in Windows

Andjii, 2021-11-24:
Hi everyone,

I've been trying to port some neural network code from a colleague that uses OpenPCDet from Linux to Windows 10, and everytime we try to execute the setup.py file, we have the same error : 

[image]
(Please note that this error even happends when installing PCDet in standalone)

Is OpenPCDet compatible with Windows? If yes, any idea on this issue?

We've tried with : 

- Cuda versions 10.2, 11.3
- PyTorch versions 1.1, 1.10
- spconv 2.1 (using the latest master commit of OpenPCDet that is supposed to be compatible with this version).
- Python 3.7.12 (via Anaconda3

xiazhongyv, 2021-11-25:
PyTorch version 1.1 fell behind, it's too old for spconv.
And spconv for cuda11.3 only support Linux.
Have you tried on:
python 3.7
Cuda 10.2
PyTorch 1.10
spconv-cu102

Andjii, 2021-11-25:
Hi! Just tried with the version you suggested @xiazhongyv. 

Most specifically : 
- Python 3.7.11
- Cuda 10.2.2
- PyTorch 1.10
- spconv-cu102
- OpenPCDet (latest master commit 70857b8)

Unfortunately I still have the exact same error
[image]

MartinHahner, 2021-11-25:
What GPU do you have? 

"undefined in device code" perhaps indicates that it might be an error related to your GPU.

Andjii, 2021-11-26:
Hi @MartinHahner !
I have a Nvidia Quadro M1000M !
I simply installed the Cuda Toolkit... Maybe there is something missing that I need to "activate" ... ?

Andjii, 2021-11-26:
From what I just tried, it seems the errors comes when compiling a certain file, trying to compile the file all by itself gives me the exact same error :
[image]

This file can be found in pcdet\ops\iou3d_nms\src\pcdet\ops\iou3d_nms\src.cu . Hopefully this could help narrow down the issue.

Andjii, 2021-11-26:
Ok, by doing some more information I found this (https://docs.nvidia.com/cuda/cuda-c-programming-guide/index.html#const-variables) : 
> The value of V may be directly used in device code, if :
>     - V has been initialized with a constant expression before the point of use,
>     - the type of V is not volatile-qualified, and
>     - it has one of the following types:
>         - builtin floating point type except when the Microsoft compiler is used as the host compiler,
>         - builtin integral type.

Checking inside this "iou3d_nms_kernel.cu" file, I noticed the problematic EPS variable is indeed a const floating variable. By trying to compile on windows, we don't follow the conditions above. I tried to modify the file by replacing the variable references directly by the value. The error does not appear anymore when compiling, but I now have other errors, here is the whole log when trying to execute "python setup.py develop" : 

```
python.exe "D:/Git Repositories/OpenPCDet/setup.py" develop
running develop
running egg_info
writing pcdet.egg-info\PKG-INFO
writing dependency_links to pcdet.egg-info\dependency_links.txt
writing requirements to pcdet.egg-info\requires.txt
writing top-level names to pcdet.egg-info\top_level.txt
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info\SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/1] cl /showIncludes /nologo /Ox /W3 /GL /DNDEBUG /MD /MD /wd4819 /wd4251 /wd4244 /wd4267 /wd4275 /wd4018 /wd4190 /EHsc -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\torch\csrc\api\include -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\TH -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\THC "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include" -IC:\Users\uie59038\.conda\envs\centerpoint\include -IC:\Users\uie59038\.conda\envs\centerpoint\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" -c "D:\Git Repositories\OpenPCDet\pcdet\ops\iou3d_nms\src\iou3d_nms.cpp" /Fo"D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src/iou3d_nms.obj" -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 /std:c++14
FAILED: D:/Git Repositories/OpenPCDet/build/temp.win-amd64-3.7/Release/pcdet/ops/iou3d_nms/src/iou3d_nms.obj 
cl /showIncludes /nologo /Ox /W3 /GL /DNDEBUG /MD /MD /wd4819 /wd4251 /wd4244 /wd4267 /wd4275 /wd4018 /wd4190 /EHsc -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\torch\csrc\api\include -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\TH -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\THC "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include" -IC:\Users\uie59038\.conda\envs\centerpoint\include -IC:\Users\uie59038\.conda\envs\centerpoint\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" -c "D:\Git Repositories\OpenPCDet\pcdet\ops\iou3d_nms\src\iou3d_nms.cpp" /Fo"D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src/iou3d_nms.obj" -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 /std:c++14
C:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\c10/util/Optional.h(184): warning C4624: 'c10::constexpr_storage_t<T>': destructor was implicitly defined as deleted
        with
        [
            T=at::TensorBase
        ]
C:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\c10/util/Optional.h(383): note: see reference to class template instantiation 'c10::constexpr_storage_t<T>' being compiled
        with
        [
            T=at::TensorBase
        ]
[... cut ...]
  File "C:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\utils\cpp_extension.py", line 717, in win_wrap_ninja_compile
    with_cuda=with_cuda)
  File "C:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\utils\cpp_extension.py", line 1404, in _write_ninja_file_and_compile_objects
    error_prefix='Error compiling objects for extension')
  File "C:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\utils\cpp_extension.py", line 1733, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension

Process finished with exit code 1

```

MartinHahner, 2021-11-26:
> Nvidia Quadro M1000M

You are using a GPU with the Maxwell architecture:
https://en.wikipedia.org/wiki/CUDA

You need to have at least GPU with the Pascal architecture:
https://github.com/traveller59/spconv/issues/152#issue-631745341

Andjii, 2021-11-26:
@MartinHahner Hmmm I'm surprised by that, I've not seen anywhere in their repo before talking about some hardware constraints, especially knowing that Maxwell architecture is supposed to be compatible with Cuda 10.2 ...

Andjii, 2021-11-26:
@MartinHahner Tried to install it in a machine with a Pascal GPU (Quadro P2000) with Windows 10, and the same issue remains.

FindDefinition, 2021-11-26:
1. This [line](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp#L116) is invalid in c++ standard, it's valid in linux because only gcc allow this syntax, see [this](https://stackoverflow.com/a/15013295). you can use ```std::vector<unsigned long long> remv_cpu(col_blocks)``` instead.
2. spconv 2.x currently don't support sm_50 GPUs because it doesn't add sm_50 to compiler flag, it support [these arches](https://github.com/traveller59/spconv#prebuilt-gpu-support-matrix). I will add sm_50 to spconv-cu102 in next minor release.

Andjii, 2021-11-29:
@FindDefinition Thanks for the heads-up! The code not being C++ standard is not ideal indeed...! I applied your suggestion to modify the code, and I also had to modify some more bit of code in Lines 118 and 169 of iou3d_nms.cpp from : 
`memset(remv_cpu, 0, col_blocks * sizeof(unsigned long long));`
to
`memset(&remv_cpu[0], 0, col_blocks * sizeof(unsigned long long));`.

Now I still have errors when trying to compile ... : 

```
python.exe "D:/Git Repositories/OpenPCDet/setup.py" develop
running develop
running egg_info
writing pcdet.egg-info\PKG-INFO
writing dependency_links to pcdet.egg-info\dependency_links.txt
writing requirements to pcdet.egg-info\requires.txt
writing top-level names to pcdet.egg-info\top_level.txt
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info\SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/1] C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\bin\nvcc --generate-dependencies-with-compile --dependency-output "D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src/iou3d_nms_kernel.obj".d --use-local-env -Xcompiler /MD -Xcompiler /wd4819 -Xcompiler /wd4251 -Xcompiler /wd4244 -Xcompiler /wd4267 -Xcompiler /wd4275 -Xcompiler /wd4018 -Xcompiler /wd4190 -Xcompiler /EHsc -Xcudafe --diag_suppress=base_class_has_different_dll_interface -Xcudafe --diag_suppress=field_without_dll_interface -Xcudafe --diag_suppress=dll_interface_conflict_none_assumed -Xcudafe --diag_suppress=dll_interface_conflict_dllexport_assumed -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\torch\csrc\api\include -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\TH -IC:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\include\THC "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include" -IC:\Users\uie59038\.conda\envs\centerpoint\include -IC:\Users\uie59038\.conda\envs\centerpoint\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" -c "D:\Git Repositories\OpenPCDet\pcdet\ops\iou3d_nms\src\iou3d_nms_kernel.cu" -o "D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src/iou3d_nms_kernel.obj" -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_50,code=compute_50 -gencode=arch=compute_50,code=sm_50
iou3d_nms_kernel.cu
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:C:\Users\uie59038\.conda\envs\centerpoint\lib\site-packages\torch\lib "/LIBPATH:C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\lib/x64" /LIBPATH:C:\Users\uie59038\.conda\envs\centerpoint\libs /LIBPATH:C:\Users\uie59038\.conda\envs\centerpoint\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x64" c10.lib torch.lib torch_cpu.lib torch_python.lib cudart.lib c10_cuda.lib torch_cuda.lib /EXPORT:PyInit_iou3d_nms_cuda "D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src/iou3d_cpu.obj" "D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src/iou3d_nms_api.obj" "D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src/iou3d_nms.obj" "D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src/iou3d_nms_kernel.obj" /OUT:build\lib.win-amd64-3.7\pcdet\ops\iou3d_nms\iou3d_nms_cuda.cp37-win_amd64.pyd "/IMPLIB:D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src\iou3d_nms_cuda.cp37-win_amd64.lib"
iou3d_nms_kernel.obj : error LNK2005: "int __cdecl check_rect_cross(struct Point const &,struct Point const &,struct Point const &,struct Point const &)" (?check_rect_cross@@YAHAEBUPoint@@000@Z) already defined in iou3d_cpu.obj
   Creating library D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src\iou3d_nms_cuda.cp37-win_amd64.lib and object D:\Git Repositories\OpenPCDet\build\temp.win-amd64-3.7\Release\pcdet\ops\iou3d_nms\src\iou3d_nms_cuda.cp37-win_ [... cut ...]
iou3d_nms.obj : error LNK2001: unresolved external symbol "public: long * __cdecl at::TensorBase::data_ptr<long>(void)const " (??$data_ptr@J@TensorBase@at@@QEBAPEAJXZ)
build\lib.win-amd64-3.7\pcdet\ops\iou3d_nms\iou3d_nms_cuda.cp37-win_amd64.pyd : fatal error LNK1120: 1 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Professional\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

Process finished with exit code 1
```

Seems like a conflict of the same function being implemented on Cuda and on CPU at the same time...

FindDefinition, 2021-11-29:
@Andjii 
First error: if both files are cpp, two functions with same signature will cause "already defined", maybe nvcc handle this problem in different way between win and linux, so you can just use a different name of that function.
Second error: don't use ```long```, ```int```, etc in any numerical computation code, use ```int32_t, int64_t, uint32_t, uint64_t``` instead. There is only one exception: ```unsigned long long```, this class maybe different with ```uint64_t``` in nvcc, some functions (e.g. atomicAdd) only accept ```unsigned long long``` instead of ```uint64_t```, we must handle this type [carefully](https://github.com/FindDefinition/cumm/blob/main/include/tensorview/dtypes.h#L145).

Andjii, 2021-11-29:
@FindDefinition Ok so I renamed the check_rect_cross function in the .cu file to check_rect_cross_cuda, and replaced all instances of "unsigned long long" to "uint64_t", and "long" to "int32_t" both in iou3d_nms.cpp and iou32_nms_kernel.cu files. Now it compiled without error! Seems like OpenPCDet has some modifications to do to be protable to Windows...!

closed 2022-03-07 by github-actions[bot]

sainttelant, 2022-05-07:
this is helpful for compiling PCDet lib in windows, thanks a lot

sainttelant, 2022-05-07:
@Andjii ,  hello, first of all, thanks for all your discussion of compiling PCDet in windows, i ' ve encountered nearly the same issues with you when installing PCDet in windows,  finally,  i got the errors of no "sys/mman.h" file found in the system,  i realised that this is only exited in Linux/Mac system, this error was caused by installing the lib of "SharedArray" when it automatically executed the commander of "pip install SharedArray",  this library is only supported in Linux or Mac,  So,  have you encounter this error when you install PCDet Lib in Windows? if so, how did you solve it?

KolyaIvankov, 2022-05-15:
@Andjii I've had a similar proble and here high-level solutions have been discussed. But digging just a little bit into what the error really is, I realized that `EPS` in your case as in mine is most likely a precision parameter, and is given by the line like that
```
const double EPS=1E-8;
```
A dirty hack with replacing `EPS` with the value `1E-9` did the trick for me. I don't know why in my case compiler refused to recognize the global variable, but well, it works.

nicolgo, 2022-06-07:
I meet the same problem and solve it, I submit my code, and hope it can help others. see [this](https://github.com/nicolgo/3DIoUMatch/commit/a6d90be8eec5653eeaf044270fd9c0872c418901)

Carolyn0822, 2022-06-08:
> 
I also have the problem (no file "sys/mman.h" ）,do you solve it now?

Carolyn0822, 2022-06-08:
> @Andjii , hello, first of all, thanks for all your discussion of compiling PCDet in windows, i ' ve encountered nearly the same issues with you when installing PCDet in windows, finally, i got the [...]

I also have the problem (no file "sys/mman.h" ）,did you solve it ?

yihuajack, 2022-07-23:
The problem still exists on NVIDIA Quadro P3200 (Compute capability version = 6.1, i.e., code=sm_61, Pascal architecture), so this issue is still not resolved. It could keep open.