open-mmlab/OpenPCDet #686: issue with test.py

wuwenxuan0226, 2021-12-02:
When I use "python test.py --cfg_file cfgs/kitti_models/pointpillar.yaml --batch_size 8 --ckpt /home/wenxuan/model_pretrained/pointpillar_7728.pth" to evaluate the pretrained model, I got error:
2021-12-02 03:21:05,767   INFO  *************** Performance of EPOCH 7728 *****************
Traceback (most recent call last):
  File "/home/wenxuan/OpenPCDet/tools/test.py", line 200, in <module>
    main()
  File "/home/wenxuan/OpenPCDet/tools/test.py", line 196, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "/home/wenxuan/OpenPCDet/tools/test.py", line 62, in eval_single_ckpt
    eval_utils.eval_one_epoch(
  File "/home/wenxuan/OpenPCDet/tools/eval_utils/eval_utils.py", line 79, in eval_one_epoch
    sec_per_example = (time.time() - start_time) / len(dataloader.dataset)
ZeroDivisionError: float division by zero

So I check the code:
    logger.info('*************** Performance of EPOCH %s *****************' % epoch_id)
    sec_per_example = (time.time() - start_time) / len(dataloader.dataset)
    logger.info('Generate label finished(sec_per_example: %.4f second).' % sec_per_example)

I think this should be:
sec_per_example = (time.time() - start_time) / len(dataloader.dataset.sample_id_list)

Am I right? If I'm wrong, can you please tell me how to fix it? Many thanks.

wuwenxuan0226, 2021-12-03:
Sorry I forget to generate the data infos so the error happened. Another issue is that the PyYAML version>5.1 will cause a error:

TypeError: load() missing 1 required positional argument: 'Loader'

So I think in 'kitti_dataset.py' line 476, it need to be changed as:
dataset_cfg = EasyDict(yaml.load(open(sys.argv[2]), Loader=yaml.FullLoader))

Many thanks.

closed 2021-12-03 by wuwenxuan0226

yimingshan666, 2022-08-16:
Hi, I met the same issue. I used your solution, but I didn't fix my problem.

grgzam, 2022-08-27:
@yimingshan666 , I think that your "cfgs/dataset_configs/kitti_dataset.yaml" file, probably points to the wrong folder that holds your KITTI data.

It is the second line of "kitti_dataset.yaml" file : 'DATA_PATH: '