open-mmlab/OpenPCDet #688: PCDetv0.5 install in windows

VsionQing, 2021-12-03:
First  SharedArray is not supported in windows
and
Traceback (most recent call last):
File "E:\PythonFile\OpenPCDet-master---updated\tools\demo.py", line 112, in
main()
File "E:\PythonFile\OpenPCDet-master---updated\tools\demo.py", line 98, in main
pred_dicts, _ = model.forward(data_dict)
File "e:\pythonfile\openpcdet-master---updated\pcdet\models\detectors\pv_rcnn.py", line 11, in forward
batch_dict = cur_module(batch_dict)
File "D:\Anconda\envs\pytorch\lib\site-packages\torch\nn\modules\module.py", line 1102, in _call_impl
return forward_call(*input, **kwargs)
File "e:\pythonfile\openpcdet-master---updated\pcdet\models\roi_heads\pvrcnn_head.py", line 149, in forward
targets_dict = self.proposal_layer(
File "D:\Anconda\envs\pytorch\lib\site-packages\torch\autograd\grad_mode.py", line 28, in decorate_context
return func(*args, **kwargs)
File "e:\pythonfile\openpcdet-master---updated\pcdet\models\roi_heads\roi_head_template.py", line 89, in proposal_layer
selected, selected_scores = class_agnostic_nms(
File "e:\pythonfile\openpcdet-master---updated\pcdet\models\model_utils\model_nms_utils.py", line 17, in class_agnostic_nms
keep_idx, selected_scores = getattr(iou3d_nms_utils, nms_config.NMS_TYPE)(
File "e:\pythonfile\openpcdet-master---updated\pcdet\ops\iou3d_nms\iou3d_nms_utils.py", line 98, in nms_gpu
num_out = iou3d_nms_cuda.nms_gpu(boxes, keep, thresh)
RuntimeError: expected scalar type Int but found Long

The above is my error message（that i have #SharedArray）

VsionQing, 2021-12-03:
`assert boxes.shape[1] == 7
order = scores.sort(0, descending=True)[1]
if pre_maxsize is not None:
order = order[:pre_maxsize]

boxes = boxes[order].contiguous()
keep = torch.IntTensor(boxes.size(0))

thresh = int (thresh)
num_out = iou3d_nms_cuda.nms_gpu(boxes, keep, thresh)

keep = torch.LongTensor(boxes.size(0))

return order[keep[:num_out].cuda()].contiguous(), None`
I replace those code in iou3d_nms_utils.py to solve this problem, but new error appear

C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [32,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [33,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [34,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [35,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [36,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [37,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [38,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [39,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [40,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [41,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [42,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [43,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [44,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [45,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [46,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [47,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [48,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [49,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [50,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [51,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [52,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [53,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [54,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [55,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [56,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
C:\cb\pytorch_1000000000000\work\aten\src\ATen\native\cuda\IndexKernel.cu:93: block: [0,0,0], thread: [57,0,0] Assertion index >= -sizes[i] && index < sizes[i] && "index out of bounds" failed.
[... cut ...]
local_roi_grid_points = self.get_dense_grid_points(rois, batch_size_rcnn, grid_size) # (B, 6x6x6, 3)
File "e:\pythonfile\openpcdet-master---updated\pcdet\models\roi_heads\pvrcnn_head.py", line 135, in get_dense_grid_points
dense_idx = faked_features.nonzero() # (N, 3) [x_idx, y_idx, z_idx]
RuntimeError: CUDA error: device-side assert triggered
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
PS E:\PythonFile\OpenPCDet-master---updated\tools>

sshaoshuai, 2021-12-05:
You could remove the dependency on SharedArray since we do not use shared memory by default. 

For the error message, I could not reproduce the error with linux env. 
(Maybe you could simply change to `num_out = iou3d_nms_cuda.nms_gpu(boxes, keep.int(), thresh)`)

VsionQing, 2021-12-05:
I'm glad to receive your reply.And,i had known this solution a few days ago. In this way, i have reproduced the demo，however, i have some problems about train. The problem that bothers me most is how to train my data set to get PTH files, and then show the effect through demo.

Recently, i have some exams, so i have little time to work on it. Look forward to communicating with you next time

Kemo-Huang, 2022-01-06:
change the code to this:    

keep = torch.IntTensor(boxes.size(0))
num_out = iou3d_nms_cuda.nms_gpu(boxes, keep, thresh)
return order[keep[:num_out].long()].contiguous(), None

VsionQing, 2022-01-06:
徐国庆，收到谢谢

jihanyang, 2022-02-08:
use --ckpt to specify a checkpoint rather than --eval all.

VsionQing, 2022-02-08:
徐国庆，收到谢谢

VsionQing, 2022-02-08:
When I try to train myself with Kitti dataset, the output is as follows

PS E:\PythonFile\OpenPCDet-master---updated\tools&gt; python train.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 1 --workers 1  --epochs 10 --extra_tag 'mydata_1'
2022-02-08 14:05:39,907   INFO  **********************Start logging**********************
2022-02-08 14:05:39,907   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-02-08 14:05:39,908   INFO  cfg_file         cfgs/kitti_models/pv_rcnn.yaml
2022-02-08 14:05:39,908   INFO  batch_size       1
2022-02-08 14:05:39,908   INFO  epochs           10
2022-02-08 14:05:39,908   INFO  workers          1
2022-02-08 14:05:39,908   INFO  extra_tag        mydata_1
2022-02-08 14:05:39,908   INFO  ckpt             None
2022-02-08 14:05:39,908   INFO  pretrained_model None
2022-02-08 14:05:39,908   INFO  launcher         none
2022-02-08 14:05:39,908   INFO  tcp_port         18888
2022-02-08 14:05:39,908   INFO  sync_bn          False
2022-02-08 14:05:39,909   INFO  fix_random_seed  False
2022-02-08 14:05:39,909   INFO  ckpt_save_interval 1
2022-02-08 14:05:39,909   INFO  local_rank       0
2022-02-08 14:05:39,909   INFO  max_ckpt_save_num 30
2022-02-08 14:05:39,909   INFO  merge_all_iters_to_one_epoch False
2022-02-08 14:05:39,909   INFO  set_cfgs         None
2022-02-08 14:05:39,909   INFO  max_waiting_mins 0
2022-02-08 14:05:39,909   INFO  start_epoch      0
2022-02-08 14:05:39,909   INFO  save_to_file     False
2022-02-08 14:05:39,909   INFO  cfg.ROOT_DIR: E:\PythonFile\OpenPCDet-master---updated
2022-02-08 14:05:39,909   INFO  cfg.LOCAL_RANK: 0
2022-02-08 14:05:39,910   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2022-02-08 14:05:39,910   INFO
cfg.DATA_CONFIG = edict()
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2022-02-08 14:05:39,910   INFO
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: training
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-02-08 14:05:39,910   INFO
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2022-02-08 14:05:39,910   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2022-02-08 14:05:39,911   INFO
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-02-08 14:05:39,911   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2022-02-08 14:05:39,911   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_poin
ts': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXT
RA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'r
andom_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-02-08 14:05:39,911   INFO
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-02-08 14:05:39,911   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-02-08 14:05:39,911   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2022-02-08 14:05:39,911   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2022-02-08 14:05:39,911   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True,
'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2022-02-08 14:05:39,911   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2022-02-08 14:05:39,911   INFO
cfg.MODEL = edict()
2022-02-08 14:05:39,911   INFO  cfg.MODEL.NAME: PVRCNN
2022-02-08 14:05:39,912   INFO
cfg.MODEL.VFE = edict()
2022-02-08 14:05:39,912   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2022-02-08 14:05:39,912   INFO
cfg.MODEL.BACKBONE_3D = edict()
2022-02-08 14:05:39,912   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2022-02-08 14:05:39,912   INFO
cfg.MODEL.MAP_TO_BEV = edict()
2022-02-08 14:05:39,912   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2022-02-08 14:05:39,912   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
[... 512 lines cut ...]
      (6): ReLU()
      (7): Conv1d(256, 1, kernel_size=(1,), stride=(1,))
    )
    (reg_layers): Sequential(
      (0): Conv1d(256, 256, kernel_size=(1,), stride=(1,), bias=False)
      (1): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (2): ReLU()
      (3): Dropout(p=0.3, inplace=False)
      (4): Conv1d(256, 256, kernel_size=(1,), stride=(1,), bias=False)
      (5): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (6): ReLU()
      (7): Conv1d(256, 7, kernel_size=(1,), stride=(1,))
[... cut ...]
epochs:   0%|                                                                                                                                                                         | 0/10 [00:00<?, ?it/s]
train:   0%|                                                                                                                                                                            | 0/3712 [00:00<?, ?it/s]

The composition format of the data set is as follows:

How should I modify my training

In addition，i have changed      pv_rcnn.yaml  
     USE_ROAD_PLANE: False

VsionQing, 2022-02-12:
I think it may be the training failure caused by this error message
KeyError: Caught KeyError in DataLoader worker process 0.
But no solution was found. I hope to get your help
In addition, this error message only needs to be configured in yaml file   USE_ROAD_PLANE: False   
KeyError: 'road_plane'
2022-02-12 15:17:17,801   INFO  **********************Start training kitti_models/pointpillar(default)**********************
epochs:   0%|                                                                                                                                                                         | 0/10 [00:05<?, ?it/s]
Traceback (most recent call last):                                                                                                                                                  | 0/3712 [00:00<?, ?it/s]
  File "E:\PythonFile\OpenPCDet-master---updated\tools\train.py", line 201, in <module&gt;
    main()
  File "E:\PythonFile\OpenPCDet-master---updated\tools\train.py", line 153, in main
    train_model(
  File "E:\PythonFile\OpenPCDet-master---updated\tools\train_utils\train_utils.py", line 86, in train_model
    accumulated_iter = train_one_epoch(
  File "E:\PythonFile\OpenPCDet-master---updated\tools\train_utils\train_utils.py", line 19, in train_one_epoch
    batch = next(dataloader_iter)
  File "D:\Anconda\envs\pytorch\lib\site-packages\torch\utils\data\dataloader.py", line 521, in __next__
    data = self._next_data()
  File "D:\Anconda\envs\pytorch\lib\site-packages\torch\utils\data\dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "D:\Anconda\envs\pytorch\lib\site-packages\torch\utils\data\dataloader.py", line 1229, in _process_data
    data.reraise()
  File "D:\Anconda\envs\pytorch\lib\site-packages\torch\_utils.py", line 434, in reraise
    raise exception
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "D:\Anconda\envs\pytorch\lib\site-packages\torch\utils\data\_utils\worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "D:\Anconda\envs\pytorch\lib\site-packages\torch\utils\data\_utils\fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "D:\Anconda\envs\pytorch\lib\site-packages\torch\utils\data\_utils\fetch.py", line 49, in <listcomp&gt;
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "E:\PythonFile\OpenPCDet-master---updated\tools\..\pcdet\datasets\kitti\kitti_dataset.py", line 424, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "E:\PythonFile\OpenPCDet-master---updated\tools\..\pcdet\datasets\dataset.py", line 127, in prepare_data
    data_dict = self.data_augmentor.forward(
  File "E:\PythonFile\OpenPCDet-master---updated\tools\..\pcdet\datasets\augmentor\data_augmentor.py", line 112, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "E:\PythonFile\OpenPCDet-master---updated\tools\..\pcdet\datasets\augmentor\database_sampler.py", line 245, in __call__
    data_dict = self.add_sampled_boxes_to_scene(data_dict, sampled_gt_boxes, total_valid_sampled_dict)
  File "E:\PythonFile\OpenPCDet-master---updated\tools\..\pcdet\datasets\augmentor\database_sampler.py", line 162, in add_sampled_boxes_to_scene
    sampled_gt_boxes, data_dict['road_plane'], data_dict['calib']
KeyError: 'road_plane'

VsionQing, 2022-03-15:
收到谢谢

closed 2022-04-29 by github-actions[bot]

Petros626, 2024-10-01:
```
###########################
1. Anaconda env mit Python#
###########################
conda create --name openpcdet python=3.8

################################################################################
#2. Deinstallation von alter CUDA Version / Installation von neuer CUDA Version#
################################################################################
https://gist.github.com/kmhofmann/cee7c0053da8cc09d62d74a6a4c1c5e4
https://medium.com/virtual-force-inc/a-step-by-step-guide-to-install-nvidia-drivers-and-cuda-toolkit-855c75efcdb6

##########################################
3. Installation OpenPCDet with Anaconda3:#
##########################################
https://medium.com/@alifyafebriana/setting-up-3d-open-source-openpcdet-with-anaconda-a-step-by-step-guide-66126107215
conda activate openpcdet
PyTorch 2: pip3 install torch torchvision torchaudio --index-url https://download.pytorch.org/whl/cu118
CUDA 11.8: siehe oben
spconv: pip install spconv-cu118
git clone https://github.com/open-mmlab/OpenPCDet.git
cd OpenPCDet
pip install -r requirements.txt
python setup.py develop
python demo.py --cfg_file /home/rlab10/OpenPCDet/tools/cfgs/kitti_models/pointpillar.yaml --ckpt /home/rlab10/OpenPCDet/pointpillar_7728.pth --data_path /home/rlab10/OpenPCDet/data/kitti/000006.bin

###################################################################
4. Download der KITTI-Daten und Erstellung der Dateninformationen:#
###################################################################
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml
```
the requirements.txt:
```txt
numpy
llvmlite
numba
torch>=1.1
tensorboardX
easydict
pyyaml
scikit-image
tqdm
torchvision
SharedArray
opencv-python
pyquaternion
av2==0.2.0
kornia==0.6.5
```