open-mmlab/OpenPCDet #699: Bug report for version 0.5

hz3014, 2021-12-06:
hi @sshaoshuai , thank you for updateing version 0.5. However, i noticed some bug when trying to train Kitti models. I believe it is related to use_shared_memory feature that you recently added.

Here is the error message:
> Traceback (most recent call last):
  File "train.py", line 201, in <module>
    main()
  File "train.py", line 104, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "../pcdet/datasets/__init__.py", line 46, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "../pcdet/datasets/kitti/kitti_dataset.py", line 23, in __init__
    super().__init__(
  File "../pcdet/datasets/dataset.py", line 30, in __init__
    self.data_augmentor = DataAugmentor(
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 23, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 27, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "../pcdet/datasets/augmentor/database_sampler.py", line 27, in __init__
    [self.db_infos[cur_class].extend(infos[cur_class]) for cur_class in class_names]
  File "../pcdet/datasets/augmentor/database_sampler.py", line 27, in <listcomp>
    [self.db_infos[cur_class].extend(infos[cur_class]) for cur_class in class_names]
KeyError: 'Cyclist'
Exception ignored in: <function DataBaseSampler.__del__ at 0x7f8148f9c1f0>
Traceback (most recent call last):
  File "../pcdet/datasets/augmentor/database_sampler.py", line 59, in __del__
    if self.use_shared_memory:
AttributeError: 'DataBaseSampler' object has no attribute 'use_shared_memory'

I tried to comment some code and bypass the error, but another error shows up:
> Traceback (most recent call last):
  File "train.py", line 201, in <module>
    main()
  File "train.py", line 104, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "../pcdet/datasets/__init__.py", line 46, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "../pcdet/datasets/kitti/kitti_dataset.py", line 23, in __init__
    super().__init__(
  File "../pcdet/datasets/dataset.py", line 30, in __init__
    self.data_augmentor = DataAugmentor(
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 23, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 27, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "../pcdet/datasets/augmentor/database_sampler.py", line 27, in __init__
    [self.db_infos[cur_class].extend(infos[cur_class]) for cur_class in class_names]
  File "../pcdet/datasets/augmentor/database_sampler.py", line 27, in <listcomp>
    [self.db_infos[cur_class].extend(infos[cur_class]) for cur_class in class_names]
KeyError: 'Cyclist'

Appreciated if you can fix the KITTI training pipeline for version0.5

jihanyang, 2021-12-06:
Have you re-built the pcdet after update to version 0.5? I can run our pcdet on KITTI normally.

hz3014, 2021-12-06:
> Have you re-built the pcdet after update to version 0.5? I can run our pcdet on KITTI normally.

Yes, i did. I kept both version 0.3 and 0.5 and rebuild based-on 0.5 repo. I will try it again.

hz3014, 2021-12-06:
@jihanyang I double checked my conda installed list and it show pcdet has been installed for version 0.5.
[image]
 But keep reporting AttributeError: 'DataBaseSampler' object has no attribute 'use_shared_memory'

sshaoshuai, 2021-12-06:
You could try to remove the `build` in this project and run `python setup.py develop` and then see whether it is good. 

If the error still happens, you'd better re-clone the project and re-build it.

hz3014, 2021-12-06:
> You could try to remove the `build` in this project and run `python setup.py develop` and then see whether it is good.
> 
> If the error still happens, you'd better re-clone the project and re-build it.

Thank you for your reply, I actually tried all your suggested methods before I wrote the issue but none of them work. I will try in docker again to see if docker helps to solve the problem.

sshaoshuai, 2021-12-06:
It's strange since I also met this problem before and it sould be an enviroment problem. 
Another way is that you could try to re-create a new conda environment + re-clone/re-build OpenPCDet.

sshaoshuai, 2021-12-06:
BTW, what's your error? The first one (shared memory) or the second one (cyclist)?

hz3014, 2021-12-06:
The first error (shared memory) should be the most direct error.
The second error happens when i commented some of the shared memory code to bypass the shared memory error.

sshaoshuai, 2021-12-06:
The first error should be the enviroment problem. 

For the second error, the reason may be that you do not have full kitti data to create the gt database, and you could change it as follows:
``[self.db_infos[cur_class].extend(infos[cur_class]) for cur_class in class_names if cur_class in infos]`` to skip this error.

hz3014, 2021-12-10:
@sshaoshuai I just finished regenerate the waymo dataset. However, I am seeing another issue at very beginning of waymo train process. I checked     infos = pickle.load(f)   was loading waymo_processed_data_v0_5_0_waymo_dbinfos_train_sampled_1.pkl and my file is 107.6mb and i checked it with code pickle.Unpickler(f) to verify that path was not empty. Not sure the root cause of this, Kindly asking for help

> 2021-12-10 13:31:44,793   INFO  cfg.TAG: centerpoint
2021-12-10 13:31:44,793   INFO  cfg.EXP_GROUP_PATH: waymo_models
Traceback (most recent call last):
  File "train.py", line 201, in <module>
    main()
  File "train.py", line 104, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "../pcdet/datasets/__init__.py", line 48, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "../pcdet/datasets/waymo/waymo_dataset.py", line 23, in __init__
    super().__init__(
  File "../pcdet/datasets/dataset.py", line 30, in __init__
    self.data_augmentor = DataAugmentor(
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 23, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 27, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "../pcdet/datasets/augmentor/database_sampler.py", line 27, in __init__
    infos = pickle.load(f)
EOFError: Ran out of input

sshaoshuai, 2021-12-21:
Sorry, I never met this problem and no idea about this error. May still need to check the environment or the process of data generation.

closed 2022-03-07 by github-actions[bot]