open-mmlab/OpenPCDet #702: AttributeError: 'DataBaseSampler' object has no attribute 'use_shared_memory' for multiple-gpu.

tdzdog, 2021-12-08:
I have recently transferred to pcdet 0.5. However, I met the above problem for multiple-gpu training (single GPU works fine). It is weird.  What is the reason?

MartinHahner, 2021-12-08:
See https://github.com/open-mmlab/OpenPCDet/issues/699

tdzdog, 2021-12-08:
> See #699

Thank for replying, Martion. But I have tried to re-build pcdet as mentioned in #699 but it does not work.

sshaoshuai, 2021-12-09:
You could re-try it carefully (create a new conda enviroment for it or remove all the previous pcdet versions), since I am pretty sure it is a problem of environment.

hz3014, 2021-12-10:
> > See #699
> 
> Thank for replying, Martion. But I have tried to re-build pcdet as mentioned in #699 but it does not work.

Simple walk around is just to comment def __del__(self): function in database_sampler.py

closed 2021-12-17 by sshaoshuai