open-mmlab/OpenPCDet #704: AttributeError: type object 'spconv.core_cc.csrc.sparse.all.SpconvOps' has no attribute 'generate_subm_conv_inds'

Karthikeyanc2, 2021-12-08:
Hi,

Thanks for the great repo. 
I am facing some issues while training pv_rcnn model (with Kitti dataset)

[image]

Spcov version: 2.1.20
pcdet version: 0.5
torch version: 1.10
cuda version: 11.3

I do not know if the error is from spconv side or from pcdet side.
Note: Pointpillars network runs perfectly.

Thanks,
Karthik

Karthikeyanc2, 2021-12-08:
Sorry, wrong spconv installation.

closed 2021-12-08 by Karthikeyanc2

LegendSun0, 2022-03-18:
> 抱歉，spconv 安装错误。

麻烦问一下，应该安装Spcov什么版本？

Stone-sy, 2022-07-03:
我遇到了和你一样的问题，请问你解决了吗？

KyleYueye, 2022-07-03:
> 我遇到了和你一样的问题，请问你解决了吗？

一样的问题……

Stone-sy, 2022-07-03:
您好！您的邮件我已收到！

brillint, 2022-07-04:
> 我遇到了和你一样的问题，请问你解决了吗？
你安装的是cpu版本，gpu版本应该这样： pip install spconv-cu113 -i https://pypi.tuna.tsinghua.edu.cn/simple/

Xavier-wa, 2023-09-25:
my device is jetson nx xavier and ubuntu ,spconv:v2.3.6 cu114,but I got the error:from spconv.core_cc.csrc.sparse.all import SpconvOps
ModuleNotFoundError: No module named 'spconv.core_cc'

Stone-sy, 2023-09-25:
您好！您的邮件我已收到！

swaggywilliam, 2023-12-04:
> > 我遇到了和你一样的问题，请问你解决了吗？
> > 你安装的是cpu版本，gpu版本应该这样： pip install spconv-cu113 -i https://pypi.tuna.tsinghua.edu.cn/simple/

I installed spconv1.2.1 as [follows](https://github.com/traveller59/spconv/tree/v1.2.1):

1. Use git clone xxx.git --recursive to clone this repo.

2. Install boost headers to your system include path, you can use either sudo apt-get install libboost-all-dev or download compressed files from boost official website and copy headers to include path.

3. Download cmake >= 3.13.2, then add cmake executables to PATH.

4. Ensure you have installed pytorch 1.0+ in your environment, **run python setup.py bdist_wheel (don't use python setup.py install).**

5. Run cd ./dist, use pip to install generated whl file.

and I change the version.txt from 2.3.6 to 1.2.1, then I get the only one wheel file in 
[image]

as you can see, what I got maybe was a CPU version spconv,  and how can I get a GPU version?
The version of spconv installed with the command you suggested is the newest version or others?