open-mmlab/OpenPCDet #705: RuntimeError: Error compiling objects for extension

JianyyuWang, 2021-12-09:
When I execute python setup.py develop
I got this error:
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/4] /usr/local/cuda:/usr/local/cuda/bin:/usr/local/cuda:/usr/local/cuda:/bin/nvcc -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/TH -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda:/usr/local/cuda/bin:/usr/local/cuda:/usr/local/cuda:/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/include/python3.8 -c -c /home/wangjianyu/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_70,code=sm_70 -std=c++14
FAILED: /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o 
/usr/local/cuda:/usr/local/cuda/bin:/usr/local/cuda:/usr/local/cuda:/bin/nvcc -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/TH -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda:/usr/local/cuda/bin:/usr/local/cuda:/usr/local/cuda:/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/include/python3.8 -c -c /home/wangjianyu/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_70,code=sm_70 -std=c++14
/bin/sh: 1: /usr/local/cuda:/usr/local/cuda/bin:/usr/local/cuda:/usr/local/cuda:/bin/nvcc: not found
[2/4] c++ -MMD -MF /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_cpu.o.d -pthread -B /home/wangjianyu/config/anaconda3/envs/OpenPCDet/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/TH -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda:/usr/local/cuda/bin:/usr/local/cuda:/usr/local/cuda:/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/include/python3.8 -c -c /home/wangjianyu/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp -o /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
FAILED: /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_cpu.o 
c++ -MMD -MF /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_cpu.o.d -pthread -B /home/wangjianyu/config/anaconda3/envs/OpenPCDet/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/TH -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda:/usr/local/cuda/bin:/usr/local/cuda:/usr/local/cuda:/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/include/python3.8 -c -c /home/wangjianyu/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp -o /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from /home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/ATen/Parallel.h:149:0,
                 from /home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/utils.h:3,
                 from /home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:5,
                 from /home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/nn.h:3,
                 from /home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/all.h:7,
                 from /home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/extension.h:4,
                 from /home/wangjianyu/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:10:
/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/ATen/ParallelOpenMP.h:84:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
 #pragma omp parallel for if ((end - begin) >= grain_size)
 
/home/wangjianyu/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:12:10: fatal error: cuda.h: 没有那个文件或目录
 #include <cuda.h>
          ^~~~~~~~
compilation terminated.
[3/4] c++ -MMD -MF /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o.d -pthread -B /home/wangjianyu/config/anaconda3/envs/OpenPCDet/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/TH -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda:/usr/local/cuda/bin:/usr/local/cuda:/usr/local/cuda:/include -I/home/wangjianyu/config/anaconda3/envs/OpenPCDet/include/python3.8 -c -c /home/wangjianyu/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_api.cpp -o /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
FAILED: /home/wangjianyu/workspace/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o 
[... cut ...]
  File "/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1228, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/home/wangjianyu/config/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1529, in _run_ninja_build
    raise RuntimeError(message)
______________________________________________________________________________________________________________________________________
my env is:
pytorch=1.6.0 cuda=10.2
spconv=spconv-cu102

guanglei-maker, 2021-12-11:
Same question.Have you solved it？

JianyyuWang, 2021-12-12:
not yet

jihanyang, 2022-02-17:
Can you import spconv normally?

JianyyuWang, 2022-03-18:
yes spconv==2.X

gashel, 2022-04-07:
I got the same question when the torch version was 0.9.0.
And I reinstall the torch with version 0.10.0，then the question was solved.
My enviroment was cuda11.1+python3.6+torch1.10.0-cu111+spconv-cu111。
When I ran the commad "pip install -r requirements.txt"，I did not install torch and torchvision. 
Instead，I installed torch with "pip install XXX.whl”

closed 2022-05-23 by github-actions[bot]

Clichong, 2022-11-09:
same question