open-mmlab/OpenPCDet #710: Cannot train with Waymo Pipeline v0.5 and 0.51, EOFError: Ran out of input

hz3014, 2021-12-13:
@sshaoshuai 
I rebuilt the repo and I finished regenerating the waymo dataset. 
However, I am seeing  issue at very beginning of waymo train process.
I checked infos = pickle.load(f) was loading waymo_processed_data_v0_5_0_waymo_dbinfos_train_sampled_1.pkl and my file is 107.6mb and i checked it with code pickle. Unpickler(f) to verify that path was not empty. Not sure the root cause of this, Kindly asking for help.

> Traceback (most recent call last):
  File "train.py", line 201, in <module>
    main()
  File "train.py", line 104, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "../pcdet/datasets/__init__.py", line 48, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "../pcdet/datasets/waymo/waymo_dataset.py", line 23, in __init__
    super().__init__(
  File "../pcdet/datasets/dataset.py", line 30, in __init__
    self.data_augmentor = DataAugmentor(
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 23, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 27, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "../pcdet/datasets/augmentor/database_sampler.py", line 26, in __init__
    infos = pickle.load(f)
EOFError: Ran out of input

hz3014, 2021-12-13:
#699

sshaoshuai, 2021-12-13:
It seems your file size is incorrect (your saving process may be interrupted for any reason), you may need to re-generate the gt database.

Here are the sizes of my generated files for your reference:
```
2.6G Nov 24 13:16 waymo_processed_data_v0_5_0_infos_train.pkl
136K Nov 24 14:23 waymo_processed_data_v0_5_0
655M Nov 24 14:28 waymo_processed_data_v0_5_0_infos_val.pkl
369M Nov 24 21:01 waymo_processed_data_v0_5_0_gt_database_train_sampled_1
1.1G Nov 24 21:02 waymo_processed_data_v0_5_0_waymo_dbinfos_train_sampled_1.pkl
15G Nov 24 21:04 waymo_processed_data_v0_5_0_gt_database_train_sampled_1_global.npy 
```

hz3014, 2021-12-13:
> It seems your file size is incorrect (your saving process may be interrupted for any reason), you may need to re-generate the gt database.
>
> Here are the sizes of my generated files for your [...]

I am not sure if anything is wrong with the generation code or the generation was interrupted. I actually generated waymo data twice and second time dataset was deleted and regenerated,  both waymo_processed_data_v0_5_0_waymo_dbinfos_train_sampled_1.pkl is sized at 107.6mb(Ubuntu FIles). However, indeed i had interruption for both data generations.

I will regenerate tonight and let you know.

hz3014, 2021-12-22:
@sshaoshuai I am sorry that things gets more complicated. I have recreated conda environment and uses spconv2.
When I tried part of raw segments (10 -20) for data generation, waymo pipeline was able to train. However, when i generated the full dataset, I am still getting EOFError: Ran out of input.

> Traceback (most recent call last):
  File "train.py", line 201, in <module>
    main()
  File "train.py", line 104, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "/media/avt/df731089-6b91-472d-9cb4-deb5213e9cb82/OpenPCDetv0.51/OpenPCDet/tools/../pcdet/datasets/__init__.py", line 50, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "/media/avt/df731089-6b91-472d-9cb4-deb5213e9cb82/OpenPCDetv0.51/OpenPCDet/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 23, in __init__
    super().__init__(
  File "/media/avt/df731089-6b91-472d-9cb4-deb5213e9cb82/OpenPCDetv0.51/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 30, in __init__
    self.data_augmentor = DataAugmentor(
  File "/media/avt/df731089-6b91-472d-9cb4-deb5213e9cb82/OpenPCDetv0.51/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 23, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "/media/avt/df731089-6b91-472d-9cb4-deb5213e9cb82/OpenPCDetv0.51/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 27, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "/media/avt/df731089-6b91-472d-9cb4-deb5213e9cb82/OpenPCDetv0.51/OpenPCDet/tools/../pcdet/datasets/augmentor/database_sampler.py", line 28, in __init__
    infos = pickle.load(f)
EOFError: Ran out of input

I think at the very end of full waymo data generation, there was something wrong:
> gt_database sample: 158079/158081
gt_database sample: 158080/158081
gt_database sample: 158081/158081
Database Vehicle: 1194368
Database Pedestrian: 1114091
Database Cyclist: 53344
Killed

WHile for part segments data generation
> gt_database sample: 3171/3172
gt_database sample: 3172/3172
Database Vehicle: 22579
Database Pedestrian: 16742
Database Cyclist: 190
---------------Data preparation Done---------------

Therefore, one file called waymo_processed_data_v0_5_0_gt_database_train_sampled_1_global.npy was not generated. Would you please look into the problem, I have spent quite few days and regenerated data quite few times. Thank you

hz3014, 2021-12-24:
I think this issue is related to RAM size, I have got 32GB RAM. Still need to test

closed 2021-12-27 by hz3014