open-mmlab/OpenPCDet #712: run python3.6 texst.py

dianbayier, 2021-12-14:
python3.6 test.py --cfg_file cfgs/nuscenes_models/cbgs_pp_multihead.yaml  --batch_size 2 --ckpt ../output/nuscenes_models/cbgs_pp_multihead/default/ckpt/checkpoint_epoch_20.pth

File "test.py", line 252, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 78, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/ll/workspace/OpenPCDet/tools/eval_utils/eval_utils.py", line 91, in eval_one_epoch
    for (i,batch_dict)in enumerate(dataloader):
  File "/home/ll/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 517, in __next__
    data = self._next_data()
  File "/home/ll/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 1199, in _next_data
    return self._process_data(data)
  File "/home/ll/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 1225, in _process_data
    data.reraise()
  File "/home/ll/.local/lib/python3.6/site-packages/torch/_utils.py", line 429, in reraise
    raise self.exc_type(msg)
FileNotFoundError: Caught FileNotFoundError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/ll/.local/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 202, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/ll/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/ll/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "../pcdet/datasets/nuscenes/nuscenes_dataset.py", line 122, in __getitem__
    points = self.get_lidar_with_sweeps(index, max_sweeps=self.dataset_cfg.MAX_SWEEPS)
  File "../pcdet/datasets/nuscenes/nuscenes_dataset.py", line 95, in get_lidar_with_sweeps
    points = np.fromfile(str(lidar_path), dtype=np.float32, count=-1).reshape([-1, 5])[:, :4]
FileNotFoundError: [Errno 2] No such file or directory: '../data/nuscenes/v1.0-trainval/samples/LIDAR_TOP/n015-2018-07-18-11-07-57+0800__LIDAR_TOP__1531883530449377.pcd.bin'

eval:   0%|                                                                                                                                 | 0/15550 [00:00<?, ?it/s]

MartinHahner, 2021-12-14:
> ../data/nuscenes/v1.0-trainval/samples/LIDAR_TOP/n015-2018-07-18-11-07-57+0800__LIDAR_TOP__1531883530449377 **.pcd.bin**

Your filename has two endings (**.pcd.bin**), check if it should be **.pcd** or **.bin**. But definitely not **.pcd.bin**.

dianbayier, 2021-12-15:
The decompression of the nuscenes data set is in this format (.pcd.bin), and the directory filename in question does not exist.

dianbayier, 2021-12-21:
The problem is solved after all the nuscenes data sets are downloaded

closed 2022-03-07 by github-actions[bot]

chandrika415, 2022-08-15:
> The problem is solved after all the nuscenes data sets are downloaded

can u please tell how did u download all the nuscenes datasets. I downloaded nuscenes mini v1.0 but i am getting same error