open-mmlab/OpenPCDet #718: Database sampler NuScenes: Sampling of consecutive (object) points even when training on single sweep

chreisinger, 2021-12-17:
The NuScenes dataset provides the training/detection on max. 10 consecutive LiDAR sweeps. OpenPCDet considers this while data extraction and gt_database generation with the flag MAX_SWEEPS. However, all sweeps may not always be required/wanted while training/inference.  Hence, if you train on less than 10 sweeps (e.g. 1 sweep), you have to extract the data with the exact number of sweeps you want. Otherwise, the data loader provides the correct point cloud (with e.g. 1 sweep), but the gt_sampler still considers all 10 sweeps and thus provides inconsistent information.

chreisinger, 2021-12-17:
I would provide a quick fix for this one, but I am not able to contribute.

MartinHahner, 2021-12-17:
@chreisinger: You can always create a pull request.

PR #719 (2021-12-17): Removes object points from gt_sampled point cloud which are not in the scope of MAX_SWEEPS
Fixes #718
Allows to train/test on NuScenes data with varying MAX_SWEEPS [1, 10] when initially generated gt_database with MAX_SWEEPS=10.
commit eed89a4d (2021-12-17): Removes object points from gt_sampled point cloud which are not in the scope of MAX_SWEEPS (#719)

* Fixes #718

* FIX: replaced wrong part
commit 1483517a (2022-01-05): Gt sampler sweeps (#721)

* Fixes #718

* FIX: replaced wrong part

* #719 Fixes index error max_dt while inference: No gt_db_sampling -> MAX_SWEEPS-1 > # unique dt (e.g. first frame)

* move max_sweeps filter to PointFeatureEncoder with new config FILTER_SWEEPS

* update to lowercase
closed 2021-12-17 by sshaoshuai